moode-player/moode #156: MPD Tweaks option for devs

bitkeeper, 2020-07-25:
When developing or testing new mpd features it isn't enough to edit the mpd.conf file, this one is overwriten by moOde. 
You need to edit the playerlib.php to include the mpd setting and lateron remove it.

### Implementing tweaks support
The proposal is to add support for a little bit easier way to tweak mpd by adding a moOde FEAT FEAT_DEVTWEAKS flag to allow mpd tweaks.

When enabled
- moOde doesn''t write the file to `/etc/mpd.conf` but to `/etc/mpd.moode.conf`.
- A script `/var/www/command/mpdconfmerge.py` is called to merge `/etc/mpd.moode.conf` and `/etc/mpd.custom.conf` into `/etc/mpd.conf`.

If the custom file doesn't exists it is ignored and the output will only contains the content of `/etc/mpd.moode.conf`.

### mpdconfmerge tool
The new mpdconfmerge tool can also be called manually to test/see the result of a merge:
```bash
pi@moodep4:~/moode/www/command $ ./mpdconfmerge.py -h
usage: mpdconfmerge.py [-h] [--dry-run] [--to-screen]
                       [--destination DESTINATION]
                       nameconf1 nameconf2

Merge MPD configuration files.

positional arguments:
  nameconf1             The name of the first configuration file. For example
                        /etc/mpd.moode.conf.
  nameconf2             Name of th configuration file to merge in the first
                        one. For example /etc/mpd.custom.conf.

optional arguments:
  -h, --help            show this help message and exit
  --dry-run             Perform a test run without writing the files.
  --to-screen           Show merged output on screen.
  --destination DESTINATION
                        Name of the merged configuration file.
```

### Example case

mpd.moode.conf:
```
log_level "standard"
bla "foobar"

resampler {
  plugin "soxr"
  quality "very high"
}
```

mpd.custom.conf:
```
log_level "verbose"

resampler {
  plugin "soxr"
  quality "very high"
  #my comment
  foo "bar"
}
```

The generated mpd.conf contains:
```
##################################################
# automatic mpd conf merge 2020-07-25 12:03:14.282893
# file 1: 'mpd.moode.conf'
# file 2: 'mpd.custom.conf'
##################################################

# setting 'log_level "standard"' is replaced by:
log_level "verbose"
bla "foobar"

resampler {
   plugin "soxr"
   quality "very high"
   #my comment
   foo "bar"
}
```

bitkeeper, 2020-07-25:
The PR #157 doesn't have yet GUI support.

moodeaudio, 2020-07-25:
Looks good. PR merged. 

Prolly no need for UI since the custom conf is hand created anyway and we update bitmask via moodeutl -A

closed 2020-07-25 by moodeaudio