moode-player/moode #158: possible issue with radio import

bitkeeper, 2020-07-25:
I could be wrong, but in `import_stations.sh` a correct restore depends on the current working directory.
Notice the destination locations of the cp commands:

```bash
...
sudo cp /tmp/var/lib/mpd/music/RADIO/*.pls var/lib/mpd/music/RADIO
sudo cp /tmp/var/local/www/imagesw/radio-logos/*.jpg var/local/www/imagesw/radio-logos
sudo cp /tmp/var/local/www/imagesw/radio-logos/thumbs/*.jpg var/local/www/imagesw/radio-logos/thumbs
...
```

moodeaudio, 2020-07-25:
That does look kind of odd doesn't it. I'll have a look.

moodeaudio, 2020-07-25:
Predictably, It bombs running it from home dir with errors corresponding to each of the var/... paths

```
cp: target 'var/lib/mpd/music/RADIO' is not a directory
cp: target 'var/local/www/imagesw/radio-logos' is not a directory
cp: target 'var/local/www/imagesw/radio-logos/thumbs' is not a directory
```
But it works fine when called from worker.php. The only thing I can think of is that the worker.php daemon must be running with / as the current directory. It's not explicitly set to / by any of the moOde sources so maybe it's something that happens when worker demonizes itself or some sort of PHP default ??

Fortunately though it will work for users but needs to be fixed for the 7 series.

moodeaudio, 2020-07-26:
A commit was pushed that fixes the paths.

closed 2020-07-26 by moodeaudio