moode-player/moode #159: ISO Image

lodgemoorlad, 2020-07-25:
Hi All
I have just been on the main website and downloaded the ISO image. Thing is, this is my first attempt to run Moode Player and I've not got past the first hurdle and I'd really appreciate a bit of help. I have tried several ways to load it onto my SD card and I'm getting nowhere at all.
Could someone tell me if this should work, just by copying it over after downloading it. Or, is there a bit I have missed. 
Also, I read this has an accesspoint built in, which is my reason for wanting to try this as I'd like a standalone system that I can control with my phone , I think its possible, but if not could someone please tell me... Thanks in advance...

moodeaudio, 2020-07-25:
moOde automatically. boots up in AP mode if no ethernet connection, no defined WiFi SSID, or it fails to get an IP address from a defined WiFi SSID. AP mode startup can take couple minutes on slow, single-core Pi's and moOde is not designed for in-car applications so YMMV.

There are some users though that have moOde rigs in their car. You might want to ask in the Moode Forum since these repos are mainly for dev type stuff.

closed 2020-07-25 by moodeaudio

lodgemoorlad, 2020-07-26:
Thanks for the quick reply Tim .
Right , got it. Good idea about the forum too. You've saved me a shed load
of frustration trying to understand moodes operating characteristics.
Thanks Alex