moode-player/moode #160: Ability to specify custom CSS

melat0nin, 2020-07-26:
It would be excellent to be able to specify custom css, saved on moode (so cross-device). I'd like to tweak the fonts etc to my taste, and while the appearance options are great some extra flexibility would be really welcome.

This could consist of a textfield stored in moOde's db, which is injected into the HTML `head` between `style` tags. Should be fairly simple to implement, I imagine (of course, it's the user's responsibility to ensure they don't enter anything that breaks the interface, although sandboxing the styles within a style tag should help avoid that).

moodeaudio, 2020-07-26:
A mechanism that allows users to patch in their own CSS would not be supportable. The best way to add flexibility to the appearance options is to propose specific new options and make a good case to have them added. Ideally they would be generally applicable to moOde users, support a desirable usage scenario or solve a specific problem.

Then if you have the expertise, fork the repo, make the code modifications and submit a PR otherwise even if the feature is one that gets added to the TODO list there is no guarantee that it's going to get priority and dev time since we are always overcommitted with feature requests.

closed 2020-07-26 by moodeaudio

melat0nin, 2020-07-26:
Not sure I agree that it's not supportable, or rather that it needs to be supported. Nobody who doesn't understand CSS will use it (and/or there could be a warning that any use is at the user's own risk). This isn't an unusual feature for a web application. But fair enough.