moode-player/moode #166: build_recipe_v2.2.6.txt - COMPONENT 5 - Squeezelite - environment variable problem

TheOldPresbyope, 2020-07-29:
While investigating the note in my issue #164 I stumbled across this problem.

Executing the lines 879-881
```
export CFLAGS="-O3 -march=native -mcpu=native -DDSD -DRESAMPLE -DGPIO -fno-fast-math -mfloat-abi=hard -pipe -fPIC"
cat ./scripts/squeezelite-ralphy-dsd.patch | sudo patch -p 0
sudo make
```
yields the unexpected output
```
cc -Wall -fPIC -O2   main.c -c -o main.o
cc -Wall -fPIC -O2   slimproto.c -c -o slimproto.o
...
```
The compiler never sees the CFLAGS value because `sudo` sanitizes the environment before executing its argument. OTOH, `sudo export CFLAGS=...` is a nonstarter.

The fix is to append the sudo option -E to preserve the variables (there are alternative approaches which would pass only CFLAGS rather than all)
```
sudo -E make
```
which yields the expected result
```
cc -O3 -march=native -mcpu=native -DDSD -DRESAMPLE -DGPIO -fno-fast-math -mfloat-abi=hard -pipe -fPIC  output_pack.c -c -o output_pack.o
cc -O3 -march=native -mcpu=native -DDSD -DRESAMPLE -DGPIO -fno-fast-math -mfloat-abi=hard -pipe -fPIC  decode.c -c -o decode.o
...
```
(To follow up on my previous note, the float-abi=hard option is accepted when I compile in 32-bit userland but not in 64-bit userland. More investigation needed.)

moodeaudio, 2020-07-29:
Try just removing sudo from the make.

TheOldPresbyope, 2020-07-29:
Nah, then you're back in permissions hell
```
pi@moode3A:~/squeezelite-master $ make
cc -O3 -march=native -mcpu=native -DDSD -DRESAMPLE -DGPIO -fno-fast-math -mfloat-abi=hard -pipe -fPIC  main.c -c -o main.o
Assembler messages:
Fatal error: can't create main.o: Permission denied
make: *** [Makefile:87: main.o] Error 2
```

moodeaudio, 2020-07-29:
Thats the recipe I used to build Squeezelite. I'll see if I can repro.

TheOldPresbyope, 2020-07-29:
Further on my note...

The [GCC ARM options list](https://gcc.gnu.org/onlinedocs/gcc-4.9.2/gcc/ARM-Options.html) includes -mfloat-abi=_name_, where permissible values of _name_ are 'soft', 'softfp', and 'hard'.

However, the [GCC AARCH64 options list](https://gcc.gnu.org/onlinedocs/gcc/AArch64-Options.html) does not include it

moodeaudio, 2020-07-30:
I created a separate 64-bit build recipe to track issues against.
https://github.com/moode-player/moode/blob/develop/other/build/build_recipe_v2.26_64bit.txt

closed 2020-07-30 by moodeaudio