moode-player/moode #167: build_recipe_v2.2.6.txt - COMPONENT 5 - Squeezelite - compiler options and arm64 build

TheOldPresbyope, 2020-07-29:
### Problem
Line 879 sets the compiler options for the squeezelite build
```
export CFLAGS="-O3 -march=native -mcpu=native -DDSD -DRESAMPLE -DGPIO -fno-fast-math -mfloat-abi=hard -pipe -fPIC"
```

As mentioned in the note in #164 and #166, this combination of options causes the compile to fail in a 64-bit build
```
pi@moode64:~/squeezelite-master $ sudo -E make
cc -O3 -march=native -mcpu=native -DDSD -DRESAMPLE -DGPIO -fno-fast-math -mfloat-abi=hard -pipe -fPIC  main.c -c -o main.o
cc: error: unrecognized command line option ‘-mfloat-abi=hard’
make: *** [Makefile:87: main.o] Error 1
```

I was misled at first by the absence of the `-mfloat-abi=hard` option in the [GCC AARCH64 options list](https://gcc.gnu.org/onlinedocs/gcc/AArch64-Options.html). This was a red herring.

It seems the problem lies rather in the `-march=native -mcpu=native` options. The compiler apparently isn't very good at figuring out the ARM processors (ref: various, not-very-definitive posts found in random Web searches).

### Toward a solution
I have successfully built squeezelite on my RPi4B "moOde64" player using these compiler options
```
CFLAGS="-O3 -mcpu=cortex-a72 -DDSD -DRESAMPLE -DGPIO -fno-fast-math  -pipe -fPIC"
```

I'm listening ATM to music streamed from a Logitech Media Server to the resulting 64-bit Squeezelite Renderer.

This isn't my recommended set of options. I need to try `-march=armv8-a` instead so the binary applies to both RPi4s and RPi3s (but I don't have a 64-bit build running on an RPi3B/B+ yet to make sure).

moodeaudio, 2020-07-29:
Heading out down Route 64, lol
Maybe close the earlier Squeezelite issue?

I just was thinking that since the 64-bit OS excludes the arm6 Pi's are we getting into supporting two builds or should we plan to EOL support for the arm6?

TheOldPresbyope, 2020-07-29:
I was thinking the prior issues stood on their own and could be closed when you resolved or rejected them. I just referenced them for continuity.

As for arm6, I'd love to see it go EOL but there are die-hard RPi Zero and RPi1 owners among your users. It's not infeasible to add a bunch of conditionals to the build recipe and mosbuild but it will get increasingly tough to support --- aka "technical debt."

moodeaudio, 2020-07-29:
We prolly have to revert the 64bit-only changes to the 2.26 build recipe cos its supposed to be for the upcoming  7.0.0 prod version of moOde which is still going to be 32bit userland and 32/64bit kernel.

It looks like there will need to be a separate 64-bit build recipe and mosbuild ??

We definitely can't support two builds so we will have to plan for EOL arm6 when the time comes.

TheOldPresbyope, 2020-07-29:
I'm not sure there's not enough data yet to say, yes, definitely, separate build recipe and mosbuild. Fortunately, we're not forced to make a decision until the RPF releases an official Raspberry Pi OS 64 Lite.

moodeaudio, 2020-07-29:
True, just experimenting now :-)

moodeaudio, 2020-07-30:
I created a separate 64-bit build recipe to track issues against.
https://github.com/moode-player/moode/blob/develop/other/build/build_recipe_v2.26_64bit.txt

closed 2020-07-30 by moodeaudio