moode-player/moode #169: MPD music services plugins

TheOldPresbyope, 2020-08-02:
MPD binaries mpd-0.21.24,  mpd-21.24_p0x3, mpd-0.22\~git, and mpd-0.22\~git_p0x3 include the Tidal and Qobuz input plugins and the Soundcloud playlist plugin.

Example
```
pi@moodeSOX:/var/local/www/mpd_versions $ ./mpd-0.22~git_p0x3 -V
Music Player Daemon 0.22~git_p0x3 (0.22~git_p0x3)
...
Input plugins:
 file archive alsa tidal qobuz curl ffmpeg cdio_paranoia

Playlist plugins:
 extm3u m3u pls soundcloud flac cue embcue

Protocols:
 file:// alsa:// cdda:// ftp:// ftps:// gopher:// http:// https:// mmsh:// mmst:// qobuz:// rtmp:// rtmps:// rtmpt:// rtmpts:// rtp:// scp:// sftp:// smb:// srtp:// tidal://
...
```

Are we actually supporting these plugins in moOde? Even if we do, the Tidal plugin is officially defunct according to current MPD documentation.

moodeaudio, 2020-08-02:
No used anywhere in moOde code. I'll remove from the compile script.

moodeaudio, 2020-08-06:
Updated binaries w/o the plugins were pushed to the repo in commit a4eb566. The compile scripts were pushed in previous commits.

closed 2020-08-06 by moodeaudio