moode-player/moode #173: Album art not displayed when playlist contains extended m3u directives

sputnik13, 2020-08-12:
When an m3u playlist is loaded with optional extended m3u directives that include the display title of the song (see https://en.wikipedia.org/wiki/M3U#Extended_M3U under #EXTINF), MPD returns the display title as Name in the currentsong command.

This breaks the assumption in https://github.com/moode-player/moode/blob/develop/www/inc/playerlib.php#L3186 where the presence of a Name field assumes that a radio station is being played.

This causes any playlist items loaded with #EXTINF populated to always return the default coverart.

moodeaudio, 2020-08-12:
Try a different logical grouping of the elements in the statement. For example.
```
// Radio station
elseif ((isset($song['Name']) || substr($song['file'], 0, 4) == 'http') && !isset($current['duration'])) {
```

moodeaudio, 2020-08-14:
My proposed logic change would not have worked. 

After testing, PR 174 accepted.

closed 2020-08-14 by moodeaudio