moode-player/moode #197: More robuste autocfg handling idea

bitkeeper, 2020-09-24:
@moodeaudio and @TookaFace about making the autocfg implementation a little bit more robust:
Adding `if` construction arround each config setting handling would explode the code. Also for each config setting also a logprint is already present.

How do you guys feel about making a more generic implementation ?
With a configuration how autocfg setting should be handled.
Generic loop can than be used to handle the checking logging and calling the actual handler.

To get the idea a proto is below:

```php
// Auto-configure settings at worker startup
function autoConfig($cfgfile) {
	...
	// above existing filling of $autocfg
	
	// Define configurationHandlers with lambda functions, 
        //   - cfgkeys define which autocfg entries are required for this block
        //  - section name is only used to print the current section
       //   - handler function to handle the configuration changes, as argument a dictionary
	$configurationHandlers = [
		array( 'cfgkeys' => ['hostname'] ,'section' => 'names', 'handler' => function($values) {
			playerSession('write', 'hostname', $value['hostname']);
		}),
		array( 'cfgkeys' => ['wlanssid', 'wlanpwd'] ,'section' => 'network', 'handler' => function($values) {
			// your implementation
		} )
	];

	// shorthand to dictionary keys
        $available_configs = array_keys($autocfg);
	$section ='';
        // central handling and checking loop for autocfg settings
	foreach ($configurationHandlers as $config) {
		$values = array();

		// print new section header
		if($section != $config['section']) {
			$section = $config['section'];
			autoCfgLog('autocfg: - '. $section);
		}

		// check if alrequired cfgkeys are present
		if( !array_diff_key(array_flip($config['cfgkeys']), $available_configs) ) {
			// is so get copie all key/value sets
			foreach ($config['cfgkeys'] as $config_name) {
				$value = $autocfg[$config_name];
				$values[$config_name] = $value;
				autoCfgLog('autocfg: '. $config_name.': ' . $value);
				// remove used autoconfig
				unset($available_configs[$config_name]);
			}
			// call handler
			$config['handler'] ($values);
		}
	}

	// check for unused but supplied autocfg settings
	if( empty($available_configs) ) {
		foreach ($available_configs as $config_name) {
			autoCfgLog('autocfg: '. $config_name.': unused configuration setting, incomplete or wrong name.');
		}
	}

```

moodeaudio, 2020-09-24:
Converting from all that in-line code to a compact loop would be great.

There will need to be a switch case to handle the configs that involve more than just updating a SQL column.

TookaFace, 2020-09-25:
Yeah, good idea, it will be more future proof

bitkeeper, 2020-09-25:
@TookaFace @moodeaudio  
I made a quick proto how this could look in PR #199 (not tested). The diff is confusing, just look at the whole function.

Reusing the configurationHandlers array, makes it also possible to generate moodecfg.txt file from a running system.
What comes close to a backup (icw the radio export).

Some notes:
- At first hand the configuration file looks like an ini file, but it isn't (which prevent to use the default php parser):
  - Comment differs from init (with ini that is ';')
  - [Sections] are not really used; each config item has an unique key
 - string values aren't surrounded by ""
- If the configfile item bluetoothname would be renamed to btname, it wouldn't not require special threatment.
- config value are not validated against supported formats and or values
  - Can this break the system ?

If you don't like it, I'm not married to it  ;-)

moodeaudio, 2020-09-25:
I looked at the function this morning and I say lets go with it but we should also change to a standard .ini format and name the file moodecfg.ini

bitkeeper, 2020-09-27:
Before testing the autoconfig reader, implementing autoconfigexporter in #199.

Is now just a function called `autoconfigExtract`.

The following code demonstrates using the extractor.
```
#!/usr/bin/php
# www/command/autoconfiggenerator.php
<?php
require_once dirname(__FILE__) . '/../inc/playerlib.php';

playerSession('open', '' ,'');
session_write_close();
$autoconfig = autoconfigExtract();
echo $autoconfig;
?>
```
More pratical is to have a download button from system configuration page.

bitkeeper, 2020-09-30:
Fixed in pr #208

closed 2020-09-30 by bitkeeper