moode-player/moode #230: Allo Piano 2.1 no longer works

nrj5k, 2020-10-18:
Hello 

I have moodeaudio setup on an RPi4 (4GB) + Allo Piano 2.1 on a Kali Reclocker.
After the update to the new version it no longer finds my sound device. I did a fresh setup and it still does not work. 
The audio mode is stuck in Dual Mono even after I save it and I get Alsa errors everytime I try to play something. 
Everything was working till I did the update. I may have been 6.4.2 or 6.5 before when everything was working,

Thank you.

moodeaudio, 2020-10-18:
I would ask Allo support if their Piano driver still works on the 5.4.y kernel branch that is used starting with moOde version 6.6.0.

nrj5k, 2020-10-18:
Got it. I'll ask them and in the meantime downgrade to 6.6.0

Thank you.

closed 2020-10-19 by moodeaudio

bimoyong, 2025-09-14:
Sorry to bring this up,

Where can I download version 6.6.0?

Thanks so much for any help.