moode-player/moode #275: HDMI output clipping with volume control disabled

JWahle, 2020-12-16:
I just tested moode via HDMI and got some subtle but sometimes very audible distortion.
My volume is set to "Disabled (0dB)".

This seems to be due to the ALSA volume being at 100% (+4dB).
If I set it to 0dB via `amixer set HDMI 0`
I get clean output.
In `alsamixer` I can only set the volume percentage to 86% (-0.01dB), so setting the volume this way is not optimal, if bit perfect output is desired.

However, at the next start, the volume is back to 100%.
Could you make sure that HDMI volume is set to 0dB and stays there when volume control is disabled?

moodeaudio, 2020-12-16:
Have a look at the "Max ALSA volume" setting in Audio Config.

moodeaudio, 2020-12-16:
We use percentage because that works just fine for setting 0dB bit-perfect volume output for most audio devices and it allows users to set a particular max percentage if they desire. It's just the Pi's built in outputs that run +4dB hot at ALSA 100%. IIRC the Pi foundation did this for a reason but I've forgotten why.

I think if ALSA generally supported setting volume using dB instead of percentage it might be worth converting to that method but unfortunately it doesn't. Just try amixer set "HDMI" -10dB

JWahle, 2020-12-17:
I just googled this and this seems to be an issue of the shell and not of ALSA.
If I insert a `--` just before the `-10dB`, everything works as expected:
`amixer -c 0 sset "HDMI",0 -- "-10dB"`
and
`amixer -c 0 sset "HDMI",0 -- "2dB"`
both work fine.

I like the first solution better, but a second possibility is to set the volume to 0dB and then go the desired amount down from there:
`amixer -c 0 sset "HDMI" "0dB" && amixer -c 0 sset "HDMI" "10.dB-"`

So this is definitely possible.

moodeaudio, 2020-12-17:
If you think your changes are generally applicable then create a PR to the repo.

JWahle, 2020-12-17:
I didn't make any changes yet, however I'll try to get it working.

closed 2021-01-12 by JWahle