moode-player/moode #277: UPNP

gerben-b, 2020-12-21:
I have Moode running with only radio's on a Raspberri Pi 3, I also have a Ubuntu based UPNP server. How can i access that from Moode? I already enabled the UPNP browser in de Moode settings but that does not seem to do anything.

moodeaudio, 2020-12-21:
Try posting your question in the support section of the Moode Forum. http://moodeaudio.org/forum

closed 2020-12-21 by moodeaudio

gerben-b, 2020-12-21:
That won't help. It is a bug.

moodeaudio, 2020-12-21:
What is the bug?

gerben-b, 2020-12-21:
I followed all possible suggestions to add my upnp source to Moode bun none worked. I enabled the upnp browser and then tried to add a upnp source to the library. When i schoose scan, it reports nothin while all other upnp clients perfectly detect my upnp (rygel) server.

moodeaudio, 2020-12-21:
This is a question better addressed in the Forum. There are users with UPnP experience that might be able to help troubleshoot.

gerben-b, 2020-12-21:
Too bad, that won't work. I'll try something else then.