moode-player/moode #280: Genius lyrics API

Stephanowicz, 2020-12-24:
Hi,

it would be a nice option to have a lyrics query for the song actually playing.
The [Genius lyrics API](https://docs.genius.com/#/getting-started-h1) is free to use and provides many songlyrics.

I have searched for an easy implementation for querying and finally created a php script. Nevertheless one still needs to install the xml-php lib `sudo apt-get install php7.3-xml`, as xpath is used for parsing the lyrics-page. For the initial request I used a function by [Allen Fair](https://gist.github.com/afair/a7c7adc52b7b49bf362935e665a87633) - I didn't change it here, but there's lots of unnecessary stuff that can be removed.

In order to use the API one needs to create a free account at  [genius.com](https://genius.com/signup_or_login) - then You need to [create a new api-client](https://genius.com/api-clients) in order to generate a client access token.
This token has to be put in the first line:

`<?php
$ClientAccessToken = '<<<PUT YOUR CLIENT ACCESS TOKEN HERE>>';
$ARTIST_=shell_exec('mpc --format %artist% | head -n 1');
$ARTIST_=str_replace("\n","",$ARTIST_);
$ARTIST=str_replace(" ","+",$ARTIST_);

$TITLE_=shell_exec('mpc --format %title% | head -n 1');
$TITLE_=str_replace("\n","",$TITLE_);
$TITLE=str_replace(" ","+",$TITLE_);

function http_request($url, $opt=array()) {
/**
 * Makes a remote HTTP request and returns the response.
 *
 * @author    Allen Fair <allen.fair@gmail.com>
 * @copyright 2018 Allen Fair
 * @license   https://opensource.org/licenses/MIT  MIT
 * @code      https://gist.github.com/afair/a7c7adc52b7b49bf362935e665a87633
 *
 * @param string  $url     The full URL endpoint: "https://example.com/endpoint"
 * @param array   $opt     An Associative Array of extended request options
 *
 *        string  'method'       one of: GET, POST, PUT, PATCH, DELETE, HEAD
 *        string  'url'          URL override
 *        string  'endpoint'     Optional endpoint appended to URL
 *        string  'user'         user name to send for HTTP Basic Authorization
 *        string  'token'        creates an "Authorization: Bearer" header
 *        string  'accept'       creates an Accept: header for the mime/type
 *        string  'password'     password to send for HTTP Basic Authorization
 *        array   'headers'      array of "Header-Name: value", ...
 *        array   'query'        name=>value pairs for the Query String
 *        array   'post'         name=>value pairs for the POST Data body
 *        array   'files'        name=>['path'=>"/path/file", 'filename'=>"name.ext",
 *                                     'type'=>"mime1/mime2"] pairs of files to send
 *        array   'cookies'      name=value pairs for cookies to send
 *        array   'http_options' name=value pairs for "HTTP context options"
 *        callable 'notifications' a stream_notification_callback for events on the stream.
 *
 * @return array  [$http_status_code, $headers_hash, $body]
 *
 * Note: sending very large files/requests must fit into memory. Chunking is not supported.
 *
 * @example
 * list($status, $headers, $content) = http_request('https://www.example.com/endpoint', array(
 *   'method'       => 'POST',
 *   'headers'      => array('From'=>'pat@mycompany.com'),
 *   'accept'       => 'application/json',
 *   'user'         => $apikey, 'password'=>'', 'token'=>$token, // Authentication
 *   'query'        => array('q' => 'PHP HTTP request'),
 *   'post'         => array('email' => 'pat@mycompany.com'),
 *   'files'        => array('path'=> './avatar.png', 'filename'=>'avatar.png', 'type'=>'image/png'),
 *   'cookies'      => array('session' => $jwt),
 *   'http_options' => array('timeout'=> 5.0, 'user_agent'=>"php_request/1.0"),
 *   ));
 */
  $http_options = array_key_exists('http_options', $opt) ? $opt['http_options'] : array();
  $headers      = array_key_exists('headers', $opt) ? $opt['headers'] : array();
  $url          = array_key_exists('url', $opt) ? $opt['url'] : $url;
  $endpoint     = array_key_exists('endpoint', $opt) ? $opt['endpoint'] : null;
  $method       = array_key_exists('method', $opt) ? strtoupper($opt['method']) : 'GET';
[... 128 lines cut ...]
         $songlyrics=$e->nodeValue;
         $found=true;
      };
      $i++;
      if($i>10){break;};
    };
    if($songlyrics!=""){
      $songlyrics=nl2br($songlyrics);
      $songlyrics=str_replace("]","]<br />",$songlyrics);
      $songlyrics=str_replace("[","<br />[",$songlyrics);
      echo $artistFound.' - '.$songFound;
      echo $songlyrics;
    }
  }
  else{echo "no lyrics found..." . chr(10);};

//
}
else{echo "please start playback..." . chr(10);}

?>
</body>
</html> `

Thanks & cheers, Stephan

Stephanowicz, 2020-12-29:
Hi,

so I played a little bit with the code and managed to get it working in the moOde UI.
[image]
[image]
[image]

I used the "quick help" and the "system info" commands as a template. 

In the footer I added this div structure after the "quickhelp":
```
<!-- Lyrics -->
<div id="lyrics-modal" class="modal modal-sm hide" tabindex="-1" role="dialog" aria-labelledby="lyrics-modal-label" aria-hidden="true">
	<div class="modal-header">
		<button aria-label="Close" type="button" class="close" data-dismiss="modal" aria-hidden="true">&times;</button>
		<h3 id="lyrics-modal-label">Lyrics</h3>
	</div>
	<div class="modal-body">
		<div id="lyrics"></div>
	</div>
	<div class="modal-footer">
		<button aria-label="Close" class="btn singleton" data-dismiss="modal" aria-hidden="true">Close</button>
	</div>
</div>
```
For the main menue I added a command like in the "sys info" link as I first didn't get it working with the "data-cmd":
`<li><a href="javascript:$('#lyrics').load('lyrics.php',function(e){$('#lyrics-modal').modal('show');}); void 0"><i class="fas fa-bars sx"></i> Song lyrics</a></li>`

For the footer menu I added a "data-cmd" in the indextpl.html:
`<li><a href="#notarget" data-cmd="song-lyrics"><i class="fal fa-bars sx"></i> Song lyrics</a></li>`

In the script-library I added data-cmd in the _// Playback ellipsis context menu_ area:
```
	else if ($(this).data('cmd') == 'song-lyrics') {
		$('#lyrics').load('lyrics.php',function(e){$('#lyrics-modal').modal('show');});
//		$('#lyrics-modal').modal();
//		('#lyrics').load('lyrics.php',function(e){$('#lyrics-modal').modal('show');});
	}
```
I finally used the one-liner as in the main menue as the other didn't seem to work... maybe because I edited the min.js directly and had a typo...

Finally I added the lyrics.php which makes a call to the geniuslyrics_v3.php wich I posted before.
The direct call of the geniuslyrics_v3.php didn't seem to work with fpm. Also the approach to copy the result in a file like in the system info didn't work as it only did so once. So I used the direct approach with passthru wich is working fine.

```
require_once dirname(__FILE__) . '/inc/playerlib.php';

//$file = '/var/local/www/lyrics.txt';
//sysCmd('php /var/www/command/geniuslyrics_v3.php > ' . $file);

$cmd = 'php /var/www/command/geniuslyrics_v3.php';

passthru('sudo ' . $cmd, $text);

//$fh = fopen($file, 'r');
//$text = fread($fh, filesize($file));
//fclose($fh);

//$tpl = 'lyrics.html';
//eval('echoTemplate("' . getTemplate("templates/$tpl") . '");');
echo $text;
```

For the icon I used the bars icon but there's also a [songlist icon in the pro version](https://fontawesome.com/icons/list-music?style=light)- [somebody made a "workaround"](https://codepen.io/Alynva/pen/ZmayNG) for the Pro-Version of the songlist icon
[image]
but at moment I didn't get it working.

So, hopefully this may be included in moOde! ;)

Cheers, Stephan

moodeaudio, 2020-12-29:
Looks good.

Please make a PR against the develop branch.

-Tim

closed 2021-01-10 by Stephanowicz