moode-player/moode #292: web UI tag presentation

jelinj8, 2021-01-05:
Moode v. 7.0.1
If a tag contains e.g. `genre=<Unknown>` or just whitespace it is shown as an empty line in tag browser (unescaped value is used). That is both inconvenient and insecure.
It is a bit difficult to fix the library (quite big) when I don't know what value is there.
Otherwise thanks for a very nice tool.

[image]

moodeaudio, 2021-01-05:
Zip up one or two of the files that have the bad tag and post or email me a download link. I'll try to repro.

jelinj8, 2021-01-05:
Hello, here they are (problematic ARTIST and ALBUMARTIST):
https://drive.google.com/drive/folders/1r0pIrtq_WkLFFhgSk47FEKLioTE1c8x-
[image]

I'll delete them once You download them.
Thans.

jelinj8, 2021-01-05:
Not really, that is what I've put there (the originals I've noticed the problem on are already fixed, so I've made some manually).
The originals contained `<Various Artists>` and `<unknown>`, this is just an example. When I connect to Moode (MPD) with M.A.L.P app from android, it is displayed correctly. It was scanned ok, but for the browser it has to be escaped (< as &lt; and so on) to show correctly, otherwise it can even break the document tree. See the screenshot with the open dev console, there is the <unknown> non-tag in document tree (that was the content of GENRE id3 tag).
[image]

moodeaudio, 2021-01-05:
Your best bet at this point would be to post in the support section of the Moode Forum and be sure to provide a download link to data that can be used to reproduce the issue.

closed 2021-01-05 by moodeaudio

jelinj8, 2021-01-06:
Hello,
I've reedited the mp3s with more ilustrational values.
It made the UI even show a working link (with a bad url, but that is not a point), empty row and so on.
The content of tags is properly encoded, correctly read by MPD, only problem is when the web UI is generated. It produces even invalid HTML, there is even a security risk in that.

I've uploaded screenshots from Mp3Tag (what I've put in the tags) and screenshots from the UI (with resulting document tree and displayed result in Chrome). Other browsers can break more, Chrome is quite resistant to HTML errors.

What I'm trying to say is that it isn't safe to just take a value (from any source, be it DB, user input or anything else) and just paste it by a template to a HTML document. It should be a quite simple fix for someone familiar with Moode's sources, just escape all external values for usage in HTML (escaping of `\ < > &` characters should be enough, maybe a newline).

That's all from me, I don't know how to explain it more clearly.

moodeaudio, 2021-01-06:
You explanation was clear but I did not see any breakage in the UI caused by the tag values in your sample files. 

After a quick look at the code its because we use element.InnerHTML to output the lists and it performs automatic conversion to HTML entities. https://developer.mozilla.org/en-US/docs/Web/API/Element/innerHTML

If there is additional validation or sanitizing of tag values we should be doing then I'll definitely look into it.

[image]

jelinj8, 2021-01-06:
Check the new files, track 1, track info. In place of artists there is an active link.
For track 3 there is an extra line between artists and album artists.
It is visible in the screenshots uploaded with the files, even the parsed DOM.
For track 2 there is a thin empty line in albumartists.
Tracka 1+2 should show `<Unknown genre>` (that is the part where the DOM is broken, it is parsed as tag "unknown" with attribute "genre" without value). That is also shown as an empty line in the list.

If you are setting values directly, you should probably use "innerText" (that should escape the values), innerHTML sets unescaped contents, even whole formatted documents.
Now it is parsed as this: `<li class="lib-entry active"><various artists></li>`, which is invalid HTML (unknown and unclosed tag).
But it should be: `<li class="lib-entry active">&lt;various artists&gt;</li>`

moodeaudio, 2021-01-06:
Sure I can look at the files again but IME its rare to see tags contain anything other than valid values. The tagging utilities make sure of this and its prolly why we haven't seen any issues come up in the Forum regarding tags containing data that breaks the DOM or tags containing malicious data.

Maybe this would be an issue in music files downloaded from untrusted sources but I suppose thats a risk the user takes.

jelinj8, 2021-01-06:
I do agree, but the `<various artists>` is a COMPLETELY VALID (and quite common) result of such tagging tool (`< and >` being used as distinction of special values) and it is showing wrong. HTML links in tags also aren't completely uncommon. Putting a `<script>` tag in an mp3 is more hacking than being useful, but also possible, e.g. when downloading stolen music.

Expecting that all input will be valid is a very bad practice in programming (believe me, I'm a developer in a big telecommunications company, if something just can't go wrong, you can bet it will). Proper escaping for languages like HTML and SQL is a must.

If Firefox is showing that OK, just try chrome or check the HTML output. The screenshots show the resulting HTML DOM as Chrome sees it after rapairs.

moodeaudio, 2021-01-06:
Sure, good practices. The issue for our project is developer bandwidth. We always have way more items on the TODO list all of which could be argued are priority 1, than we have dev time to work on them.

You are welcome to join the project and go through the code :-)

jelinj8, 2021-01-06:
That is why I've created the ticket, I haven't expected instant solution. Maybe I'll try it someday, but MoOde is just a small part of my setup, I have to get rid of the 6 remote controls first. Tickets (issues) can wor quite well as a TODO list :-)

jelinj8, 2021-01-06:
OK, this really isn't escaped at all :-/.
I'll try to fix that and send a PR.

`
for (var i = 0; i < allGenres.length; i++) {
		output += '<li class="lib-entry'
			+ (LIB.filters.genres.indexOf(allGenres[i]) >= 0 ? ' active' : '')
			+ '">' + allGenres[i] + '</li>';
	}
`

jelinj8, 2021-01-06:
Here is my (probably incomplete) attempt to fix it on the most obvious places:
https://github.com/moode-player/moode/compare/develop...jelinj8:develop
https://github.com/jelinj8/moode

I have no way to test it as I'm not going to set full build environment, this done just in notepad. As I don't know the whole system, I'm not going for more attempts.

The .html("") in JQuery is exactly the opposite - set content without escaping (otherwise the surrounding HTML wouldn't work).

Good luck, I'll probably stay happy with fixing my Id3tags for now and an android app to control MPD when not near a PC.