moode-player/moode #294: Scaling problem for maxed window on windows desktop

Stephanowicz, 2021-01-05:
Hi there!

Just realized that there seems to be a problem with scaling on windows desktop - this happens when the browser window is maxed:
[image]

if I scale the view to about 60% the missing parts become visible:
[image]

this is not a browser problem as it happens  in firefox and edge chromium (ok - maybe both have the same problem xD - no)

Hope this is not a big thing...

Cheers, Stephan

moodeaudio, 2021-01-05:
I would post in the Moode Forum to see if someone with Windows can repro. Provide info on screen res, viewport size and if any Browser zoom is active.

closed 2021-01-05 by moodeaudio

Stephanowicz, 2021-01-06:
I've digged a little deeper:
It is related to the font scaling settings of windows os - as I have a small 11" display with high resolution (1920x1080) the font is scaled to 150% systemwide - and I don't wanna change this for obvious reasons.

I looked a bit into the style definitions for the relvant divs like "playback-control" but it wasn't obvious for me on first glance how the contents get scaled...

nevertheless I found a few settings that would help to display the whole content - 
my favourite is 
`#playbtns, #playbtns .btn-group {font-size:1em;margin:1em 0;}`
in panels.css
with either removing the font-size setting or setting it to "0" all contents are displayed properly.

Well I can help myself with this - but I'm sure that anyone having the system fonts scaling active will have the same problem.

moodeaudio, 2021-01-06:
The problem is that AFAIK there is no way to know in CSS whether user has applied OS-level scaling in their display. We would need to know that in order to make a CSS media query that could adjust the elements.

Stephanowicz, 2021-02-09:
Just found this:
window.devicePixelRatio does the job - well, it's not css...

[image]

Stephanowicz, 2021-02-09:
oh my... xD

[image]

monkeypox, 2021-02-09:
Can you post full screenshots before/after?

monkeypox, 2021-02-09:
Also is this a browser font size setting or the windows display scaling?

Stephanowicz, 2021-02-09:
> Can you post full screenshots before/after?
> 
> also is this using browser font scaling or the windows display scaling?

Hi - ahm "before/after" ? :D 
Didn't solve it, yet - just found those 2 ways to find out the scaling of the system (as Tim mentioned that there might be no way to get the settings)

and as mentioned before - this is a problem with OS' scaling of fonts - in my case it's 150%

monkeypox, 2021-02-09:
You said

>nevertheless I found a few settings that would help to display the whole content

We can’t replicate your display hence the request for full screenshot. The change you proposed only changes the vertical margin for the buttons a bit, I was curious to see how this affected what was shown.

How hidpi usually works is the system sees a smaller viewport than the screen resolution and automatically adjusts the font size and other layout elements to improve the quality rather than display more information, e.g. an iPad might have a 2048x1536 screen but moode sees a 1024x768 device. If your device shows a 1080p viewport that enables some different css meant for large screen displays.

Stephanowicz, 2021-02-10:
Ah - ok.
Well, this was just a workaround for me...

I think the correct way would be to integrate `@media (resolution: 150dpi)` somehow or the `window.devicePixelRatio` 
I dunno much about calculations in css - have to take a closer look - but I think it should be possible to add `resolution` to the resize settings as an additional factor.