moode-player/moode #301: Missing autocfg options

bitkeeper, 2021-01-13:
This issue is created to provide an overview of the settings still to implement for autoCfg:

System Config Page:
* System modifications
  * Kernel architecture
  * USB auto-mounter
  * US UAC2 fix
* Local Display:
  * Mouse cursor
  * Screen blank
  * Wake dispay on play
  * Brightness
  * Pixel Ratio
  * Rotate screen
* Local Service:
  * Metadata file
  * LCD update engine
  * GPIO butto handler
     * + configuration
  * SHH term server
*Maintenance:
  * Debug logging

LibraryConfig Page:
* Music Database: 
  * Update MPD database on USB inser or remove
  
Audio Config Page:
* I2S Audio Device
  * Driver Options
* Alsa Volume:
  * Max Alsa Volume
* MPD Options:
  * General:
    * MPD version:
    * Autoplay after start
  * Auto-shuffle:
    * Auto-shuffle
	* Mode
	* Tag filter
  * Volume options:
    * Volume step limit
	* Max MPD volume
	* Display dB volume
	* USB volume knob
	* Rotary encoder
  * DSP options:
    * Crossfade
  * HTTP streaming:
    * Streaming server
	* Port
	* Encoder
  * Equalizers:
    * CamillaDSP
	* Graphic EQ config from edit button)
	* CamilladDSP config from edit button)
  * Audio Renders:
	* Bluetooth:
	  * Speaker sharing
	  * BlueZ config (By edit button)
  * Airplay renderer:
      * Airplay config (By edit button)
  * Spotify renderer:
      * config (By edit button)
  * Squeezelite renderer:
      * config (By edit button)
UPnP/DLNA:
  * UPnP config (By edit button) [Here is still some Qobuz stuff left; should this be removed ?]
  
MPD Config Page:
  * DSD support	
	
Graphic EQ Curves Page:
  * Master Gain
  * Curves
	
CamillaDSP settings Page:
  * Default device

Clock Radio Settings Page:
  * _everything_
  
Input Source Page:
  * _everything_
  
Radio Manager Page:
  * Sort tag
  * Group method
  * Show/Hide:
    * moOde stations
	* other stations

moodeaudio, 2021-01-15:
Ugh, lotta work.

bitkeeper, 2021-01-15:
Yep, but for now I think it missing to much to call it a backup and restore solution.
I don't expect that everything will be implemented in one or two minor version.

For now I'm implementing the items marked with WIP.
I don't use any other renderer, so it would be nice if someone else could pickup those ;-)

moodeaudio, 2021-01-15:
Agree. I'll do the renderers.

bitkeeper, 2021-01-15:
First batch is ready in PR #307.

bitkeeper, 2021-07-09:
Also missing are the new settings from 7.3among which:
- alsa output mode
- alsa loopback

moodeaudio, 2021-10-21:
Issue has become stale but for sure updating autoconfig and backup/restore is important. How can it be prioritized is the question...

moodeaudio, 2024-04-22:
All settings in upcoming moode 9 should be reflected in autocfg. If any are missing users can report in the Forum and they will go on the TODO list.

closed 2024-04-22 by moodeaudio