moode-player/moode #302: Improve radio backup and restore

bitkeeper, 2021-01-13:
The current implementation is based on:
- sql dump + scheme of radio table
- mpd radio pls
- images full + thumb

Espcially the first one is an issue; it locks it against a specific db scheme.

More robust would be to generate datafile which combine the info of radion record + the info of the pls.

The import can then, in the style of autocfg,:
- import the pieces that are supported
- ignore that are nog longer supported
- convert fields that are changed
- create default values for new fields

moodeaudio, 2021-01-15:
It was mainly a check against importing incompatible 6 zips into 7 but for sure we should adopt your proposed changes going forward. There is already a new "home_page" column in cfg_radio for 7.1.0 so the import needs to be modified so it doesn't reject 7 zips.

TheOldPresbyope, 2021-01-15:
The UPnP cover art issue distracted me from finishing a mod to import_stations.sh which is intended strictly to deal with a moOde 7.0.1 player importing either a zip exported from moOde 6.7.1 or one exported from moOde 7.0.1. In the latter case, of course, it simply slurps in the zip. In the former case, it

1) fills out the missing geo_fenced, reserved1, reserved2 fields with "No","",""
2) creates new ./thumbs/<thumb>_sm.jpg files

IMO, this is a stopgap measure but it addresses the current issue.

Longer term, I believe it should be possible to select whether to export all stations or only user-defined stations. On import, it should be possible to select whether to import all stations or only user-defined stations and, in the latter case, whether to clear and add (as is done now for all stations) or to append only "new" user-defined stations. The export file should be explicitly tagged by the version of moOde which wrote it. Exporting the schema file helps, but that's indirect evidence.

moodeaudio, 2021-01-15:
I'd hold off a bit on import_stations.sh until I push a commit that adds a small change to it.

bitkeeper, 2021-01-30:
@TheOldPresbyope would you be willing take a look at the python script in #326 and provide some feedback about it ?
This would be a python script that provide most you mentioned above.
For the files it generates/imports I would like to use the extension .mrb (moOde radiostation backup) .

Not ready tested with older versions, but this single script should also work with older version of moOde.
And also import between different versions.

```
pi@moodep4:~/moode/www/command $ sudo ./stationmanager.py --help
usage: stationmanager.py [-h] [--version]
                         (--import | --export | --clear | --compare)
                         [--scope {all,moode,user}] [--how {clear,merge}]
                         [databasefile] backupfile

Manages import/export of moOde radiostations.

positional arguments:
  databasefile          File name of the database.
  backupfile            Filename of the backupfile.

optional arguments:
  -h, --help            show this help message and exit
  --version             show program's version number and exit
  --import              Import radio stations from backup.
  --export              Export radio stations to backup.
  --clear               Clear radio stations.
  --compare             Show difference between database and backup.
  --scope {all,moode,user}
                        Indicate to which stations to apply the action.
  --how {clear,merge}   On import clear stations before action or merge and
                        add.
```

TheOldPresbyope, 2021-01-31:
@bitkeeper 

I'll be happy to. It might take me a day or so---we're getting our first real snow storm and the neighborhood management group gets huffy if I don't immediately clear off my share of the sidewalks and parking spaces.

TheOldPresbyope, 2021-02-02:
@bitkeeper

Maybe I'm just not awake yet but I didn't find an Issues section on your GitHub repo so here's my first comments here.

### substantive
The script appears to work as intended when exporting from either a moOde 7.0.1 player or a moOde 6.5.2 player I spun up for the purpose.

However, whether I try to import a backup from either player to a moOde 7.0.1 player, I get an sqlite3 error
```
pi@moode:~ $ sudo ./stationmanager.py --import --how clear moode701.data
moOde database location is '/var/local/www/db/moode-sqlite3.db'
Radio logos location is '/var/local/www/imagesw/radio-logos'
MPD Radio playlists location is '/var/lib/mpd/music/RADIO'
Using moOde backup file 'moode701.data'.
import
clear stations for scope 'all'
Import 1.FM - Blues Radio
...
Import Zen FM
Traceback (most recent call last):
  File "./stationmanager.py", line 409, in <module>
    mgnr.do_import(args.scope, args.how)
  File "./stationmanager.py", line 209, in do_import
    cursor.execute(query)
sqlite3.Warning: You can only execute one statement at a time.
```

I'll look more closely at this database call later today.

### Nitpicks

- Since the extensive help output shows this script is intended to be used from the CLI as well as from within moOde, I'd suggest it explicitly warn that root privileges are needed to import a backup file.
- Since the positional argument `databasefile` is actually an optional argument I'd suggest making it so explicitly. I'm not a fan of a polymorphic script where one must remember the difference between `./stationmanager.py filea` and `/stationmanager.py filea fileb`
- Some of the warning/error messages could use a little massaging to make them more idiomatic English, but I'll save this until after the substantive issue is resolved.

### Final observation

`stationmanager.py` has all the features I envisioned and more. It's a pleasure to see Python used skillfully instead of as a Perl look-alike! Your code reminds me that I'm a jack of all trades and master of none.

bitkeeper, 2021-02-02:
@TheOldPresbyope thanks for taking a look at it.

I have updated the python file with:
* Fix for sqlite warning about multiple statements
* Make the db argument explicit 
  * (or we can also make it less flex by removing the argument and use a fixed path to the moode db, what do you prefer)
* The root privileges:
  * The help contains a message about the root privileges
  * When doing an import or clear it will check if it has root privileges, if not show a message and quit.

TheOldPresbyope, 2021-02-03:
- OK, so with this revision we get to a different database entry error caused by a failure to protect single-quote marks when constructing the query to build the table row. Here for example, I exported the moOde 7.0.1 stations then tried to import them. This failed on the first offending station,  Absolut Relax, with genre Pop, 80's, 90's. 

A snippet from the script output where I've inserted a print statement just after the query is constructed (ca line 220)
```
sudo stationmanager.py --import  --scope=all /var/local/www/db/moode-sqlite3.db moode701.data
moOde database location is '/var/local/www/db/moode-sqlite3.db'
Radio logos location is '/var/local/www/imagesw/radio-logos'
MPD Radio playlists location is '/var/lib/mpd/music/RADIO'
Using moOde backup file 'moode701.data'.
import
...
Import Absolut Relax
---query=INSERT INTO cfg_radio (station, name, type, logo, genre, broadcaster, language, country, region, bitrate, format, geo_fenced, reserved1, reserved2) VALUES ('http://stream.absolutradio.de/relax/mp3-160/radioplayer/', 'Absolut Rela
x', 'r', 'local', 'Pop, 80's, 90's', 'Absolut', 'German', 'Germany', 'Europe', '160', 'MP3', 'No', '', '');
    mgnr.do_import(args.scope, args.how)
  File "./stationmanager.py", line 200, in do_import
    self.create_query_station_to_db(station, fields_db)
  File "./stationmanager.py", line 224, in create_query_station_to_db
    cursor.execute(query)
sqlite3.OperationalError: near "s": syntax error
```

I haven't looked to see what other fields might have problematic punctuation in their entries.

- I didn't express myself clearly in my original comment about the `databasefile` argument. I was thinking it would make more sense to make this an optional argument since in most cases users will be importing/exporting to/from /var/local/www/db/moode-sqlite3.db. Rather than force the user to type this explicitly every time, why not have

`[--databasefile databasefile]`

where I couldn't think of a better name for the value, hence the redundancy:)

bitkeeper, 2021-02-03:
Oeps totally forgot to escape the sqlite query, fixed it. Every value is now surround by an StationManager.escape function.
Gues too spoiled with the frameworks that normally do all the heavy lifting for me ;-)

Ok changed databasefile to an optional named argument with the name --db. Default as with the optional argument was; /var/local/www/db/moode-sqlite3.db.

bitkeeper, 2021-02-03:
Also a filter on type is added, so you can export for example the favorites or only visible stations.

TheOldPresbyope, 2021-02-03:
Nice!

Now exports and imports the curated radio stations from 7.0.1 to 7.0.1. Will explore other combinations later.

BTW - meant to compliment your use of ffmpeg. I had my own "totally forgot" moment which meant my own script forced users to load the Python Imaging Library. Sigh.

TheOldPresbyope, 2021-02-04:
@bitkeeper 

Umm. I spoke too soon. There's a problem with problem logic here but I haven't sorted the symptoms yet. More tomorrow.

TheOldPresbyope, 2021-02-05:
@bitkeeper 

With a moOde 7.0.1 player I can export and import "all", "moode", or "user" stations and, of course, clear stations. Very nice. 

I note the following

- You and I may call them "user" stations but the moOde WebUI refers to them as "other".
- Roundtripping reveals an interesting quirk: "moode" stations become "user" stations on import, e.g., they are assigned ids > 499. To a casual user this may not be noticeable but it diminishes the utility of the hide function in the WebUI's Radio Manager.
- In moOde 7.x Tim has used "OFFLINE" as a station name (see, e.g., all the Audiophile stations) in the database, I presume to keep the ids straight for stations he believes will return to the air. The corresponding .pls file contains the last-known station URL as `FILE1=<URL>`. Upon re-import using stationmanager.py, a new .pls file is generated on the fly with `FILE1=OFFLINE`. I'm not sure this makes any difference given that we currently can't access the station via the WebUI's Station Edit popup to replace "OFFLINE" with a new URL.
- It seems ungainly to have to include an arbitrary filename in a invocation of `./stationmanager.py --clear`. I understand how using the Python argparse library forces this behavior and I see that my favorite try/except block would be difficult to use here, but still. Fortunately, a real file doesn't have to be named here, not even /dev/null; any string will do.

With moving stations from a moOde 6.x player to a moOde 7.x player some other quirks emerge, but I'll mention those in a different comment.

bitkeeper, 2021-02-05:
@TheOldPresbyope I am glad you are willing to test it so extensively.

About the points:
>You and I may call them "user" stations but the moOde WebUI refers to them as "other".
Ok I will rename user to other.

>Roundtripping reveals an interesting quirk: "moode" stations become "user" stations on import, e.g., they are assigned ids > 499. To a casual user this may not be noticeable but it diminishes the utility of the hide function in the WebUI's Radio Manager.
I think that was a problem, but was fixed yesterday in commit 79aaccf0df409195b3ab1510637c1854453c3970.

>In moOde 7.x Tim has used "OFFLINE" as a station value (see, e.g., all the Audiophile stations) in the database,
I think it hurts nobody if we skip the stations with OFFLINE as value.  I will also add that to the script.

>It seems ungainly to have to include an arbitrary filename in a invocation of ./stationmanager.py --clear. 
Yes that was one where I was also struggling with. I was in doubt if I needed to make the backup argument optional and throw out an exception if it is missing at import, export and compare. What do you prefer ?

I'm curious what still needs to be fixed for 6.x. And even 6.x to 7.x or the other way arround.

bitkeeper, 2021-02-05:
Point 1,3 and 4 are committed to repo. 
I changed the backfile to optional with an error when needed and missing. Let me know if you like this more then previous attempt.
I still hope that point 2 was already fixed yesterday.

TheOldPresbyope, 2021-02-06:
I thought I'd pulled the latest script before I started, sorry. In any case, I pulled this morning and you've nicely addressed my points. I like as well the fact that now invoking `stationmanager.py` without any arguments gets one the normal help output and no error message about missing file.

Specifically concerning point 3, I agree with your solution. However, the select statement will need also to skip "DELETED" stations which was Tim's marker in older versions of moOde. 

Concerning `import` and also your new `--type option`, note that the type values changed between 6.7.1 and 7.x, Here's a snippet from the comment in my own loadradios7.py script

```
#    'type' (6.7.1) is 'u' for user-defined or 's' for system-defined 
#    'type' (7.0.0) is 'r' for regular, 'f' for favorite, or 'h' for hidden 
#    'logo' (6.7.1) is a filepath or 'local'; user-defined stations appear
#           always to have logo='local' 
#    'logo' (7/.0.0) appears always to be 'local'
```

I grabbed the 'logo' comment lines too but I think they may be a moot point.

---

Why am I testing extensively? Three reasons mainly

- I appreciate all the hard work which has gone into moOde and I want to do what I can to be sure new features work as advertised. At work, I was the guy who could reliably break other people's software 'cuz I could think like a user who didn't know enough not to try something </brag>
- I'd already made my own attempt at dealing with station save/restore but your vision and your skills are clearly better than mine so I've switched focus
- it gives me something to do while I sit at the keyboard watching the clock for my next chance at registering for a Covid-19 vaccination. I'm like the old guy standing in front of the local OTB waiting for it to open.

bitkeeper, 2021-02-06:
@TheOldPresbyope and again you versionis outdated ;-) :
In the PR #326 I have also added support for the old backup format. 7.10 has a changes scheme for cfg_radio which would break the existing backups. Concerns support for both import and compare.

TheOldPresbyope, 2021-02-06:
In the words of my grandmother, "you've been a busy bee!"

bitkeeper, 2021-02-06:
And the DELETE is also present .. a few minutes ago ;-)

bitkeeper, 2021-02-06:
> ```
> #    'type' (6.7.1) is 'u' for user-defined or 's' for system-defined
> #    'type' (7.0.0) is 'r' for regular, 'f' for favorite, or 'h' for hidden
> #    'logo' (6.7.1) is a filepath or 'local'; [...]
Kent is the conclusion below correct ?:
* The moode vs other|user is filtered based on if the id < 499 or >499. The flags are ignored. So I think that is save.
* u and s should be dropt on import if version >7
* r,f and h should be dropt on import if version < 7
* logo should be set to 'local' if version >=7

TheOldPresbyope, 2021-02-06:
1. Yes I think so too. The presence of the 499 marker seemed to render the s|u type irrelevant but I don't know the history.
2., 3. Dropping the values going up-version certainly seems correct; I think the reverse is probably true but I'll check.
4. That's the point I'm unclear about. I haven't paid much attention to the special case of a filepath. I need to check some old distros today. If memory serves, by the time of moOde 6.5.2, only some bit-rate BBC station entries included a filepath and no user-defined stations did. I have to check to see what moOde 7.x does when it runs into a filepath.

bitkeeper, 2021-02-06:
@TheOldPresbyope 
Implemented 2 and 3 .... but to be honest didn't test it with an old db, because my time is up for today. 
If 4 is needed; you can uncomment the block at line 241-243

New version is pushed.

TheOldPresbyope, 2021-02-07:
Not to worry about testing; only one of us needs to get exhausted doing it;)

- I must have been asleep yesterday. Now that Tim uses the type value to control visibility of stations we should convert "s|u" to "r" uniformly (line 240) when moving stations up-version. Works a treat here.

- Setting the type value to null when moving stations down-version seems to do no harm (at least I found no code in moOde 6.5.2 which would get heartburn over it). However, if one goes (7.0.1)->(6.5.2)->(7.0.1) then the null value again affects visibility. I can think of three fixes: 1) when down-versioning, use id<499 to set type="s" and >499 to set type="u" (the purist solution); or 2) when up-versioning, include null as one of the set of type values to be converted to "r" (the pragmatic solution); or 3) ignore it on the assumption that nobody but me is crazy enough to have done this multi-version roundtripping (the realist solution!). 

- Regarding the logo value, I found an interesting phenomenon. If we retain the filepath value, then the stations in question display the default moOde logo in moOde 7.0.1 (because the logo file isn't found) but if the value is forced to 'local' then they display the broken-image icon. Why? Because when up-versioning, no logo image has been copied so the file your code creates contains zero bytes. 
 
The issue here is that in the past Tim apparently took advantage of the filepath vs "local" distinction [1] to fiddle with names. In particular, the roots of the image filename and of the .pls filename are the same while the NAME1= value in the .pls file and the station name value in cfg_radio are the same. Down-versioning from 7.0.1 to 6.x is not a problem because every station is logo=local so the same name is used throughout. As I think I said before, it's only the BBC 320K stations and Tim changed their naming scheme in 7.x so now everything is logo=local. An example is "BBC 320K Asian Network" in the filenames vs "BBC Asian Network - 320K" as values for 6.5.2 but "BBC Asian Network (320K)" everywhere in 7.0.1.

And why am I testing with 6.5.2? Because most of the code is the same as in the 6.6/6.7 series but as you already know some of the directories aren't.

[1] ISTR that in an early version of moOde this construction was used to refer either to an image on a station's website or to a local image. I could be wrong.

bitkeeper, 2021-02-07:
Ok clear so we have:
- type 6.x > 7.x 's|u' to 'r'
- type 7.x > 6.x id> 500 'u' else 's'
Both changes are committed.

I can see if I can fix for the logo's that aren't local but have a path by:
- importing the image by path
- writing it in the backup with the station name instead
- setting the logo to local

The end result in the backup will then only have 'local' logos. 
Will do that tomorrow.

Do you think that after that we are good to go ?

TheOldPresbyope, 2021-02-07:
That's gotta be close to perfection but after my oversight yesterday I'll hedge my bet until tomorrow:)

TheOldPresbyope, 2021-02-08:
Ugh. Where I said the NAME!= line in the .pls file, I meant to say TITLE1=
Fortunately it doesn't affect what you're proposing to do, but I edited the comment in case someone else might stumble over it.

bitkeeper, 2021-02-08:
@TheOldPresbyope push new commit with the logo!=local fixes.

It everything is right we should be able to:

- backup/restore from 6.x.x to 7.x.x
- backup/restore from 7.x.x to 6.x.x
- also support the 7.1.0 scheme changes
- import of the old backup format
- single script no need to install other deps
- scope attribute works well with both 6.x.x and 7.x.x.

Possible todo's after merge PR:
- change the messages to correct English
- add verbose argument to default suppress most messages

TheOldPresbyope, 2021-02-08:
Wow, the old backup format too!

Regardless of the backup file, though, I'm running into an issue with the station logo not appearing in the Playback view. I need a few minutes to track that down.

bitkeeper, 2021-02-08:
The PR #326 is cleaned up with a rebased and squashed the commits to a single one.
To get the latest version you might require a git pull --force on it.

If more changes are required to fix cross version backup and restore I will open a new PR.

closed 2021-02-18 by bitkeeper