moode-player/moode #308: build_recipe.txt Component 6 - Upmpdcli: prep for new cover art script

TheOldPresbyope, 2021-01-15:
First a couple of editorial issues in build_recipe.txt commit https://github.com/moode-player/moode/commit/f3dfdcbed78cf720fa9a0c8f6472c9c73789589e:

line 997 - should read Upmpdcli 1.5.8

lines 1010/1011 - /mnt/moode-player/GitHub/... Is this just a cut-n-paste that needs fixing or is the build recipe moving to a new way of doing business?

---

Now, prep for new cover art script. 

Jean-Francois has written a Python3 script, albumart.py, to replace upexplorer (which has problems with OpenHome services). To use the new script we need his Python bindings to libupnpp

Background reading: [A Python interface for libupnpp](https://www.lesbonscomptes.com/upmpdcli/libupnpp-python/upnpp-python.html)

As I mentioned to JFD by email I was unable to build the bindings on a moOde 7.0.1 system directly from his git repo following his instructions. However, I was able to build them from the source code tgz file on his downloads page [libupnpp-bindings-0.20.1.tar.gz](https://www.lesbonscomptes.com/upmpdcli/downloads/libupnpp-bindings-0.20.1.tar.gz) 

Assuming this file is already in the ./moode/other/upmpdcli directory

```
cd
cp ./moode/other/upmpdcli/libupnpp-bindings-0.20.1.tar.gz ./
tar xfz libupnpp-bindings-0.20.1.tar.gz
cd libupnpp-bindings-0.20.1
./configure --prefix=/usr PYTHON_VERSION=3
make
sudo make install
cd
rm -rf ./libupnpp-bindings-0.20.1
```

I will deal with the albumart.py script in a separate issue.

moodeaudio, 2021-01-15:
Can you post a download link for libupnpp-bindings-0.20.1.tar.gz ?

TheOldPresbyope, 2021-01-15:
Sorry, I should have made it more obvious that the highlighted name at the end of "on his downloads page..." contains the link.

moodeaudio, 2021-01-15:
Ok got it.

moodeaudio, 2021-01-15:
I noticed that JFD's recipe includes running autogen before configure
https://www.lesbonscomptes.com/upmpdcli/libupnpp-python/upnpp-python.html#_building_git_code_and_installing

I'm not sure what autogen does but should we add it to our recipe?

TheOldPresbyope, 2021-01-15:
I found it not to necessary when building from the tgz souces.

moodeaudio, 2021-01-15:
K I'll give it a go.

moodeaudio, 2021-01-16:
I compiled the lib and all looks good after testing against a simple UPnP config using miniDLNA and Linn Kazoo Ctl point. Commits have been pushed so go ahead and close when convenient.

closed 2021-01-16 by TheOldPresbyope