moode-player/moode #311: UI-Enhancements: Track-Info for PL items, Browse to folder, Multiple Albumart

Stephanowicz, 2021-01-17:
Hi,
this is intended as a suggestion for little enhancements of the UI
1. Track-Info for playlist-item:
[image]

accessible in the context-menu of a playlist-item

needs extension of search code in moode.php - btw. I think mpc has changed the search query? 
(https://linux.die.net/man/1/mpc)
`search <type> <query> [<type> <query>]...
Searches for songs where all of the given tag <type>s match the given <query>s. Any number of tag type and query combinations can be specified. Possible tag types are: artist, album, title, track, name, genre, date, composer, performer, comment, disc, filename, or any (to match any tag).
find <type> <query> [<type> <query>]...
Same as search, but match <query>s exactly.`

but in playerlib.php "search any" and "search" is used....

so I added a 'find' query to get the filepath of the corresponding song in the pl:
`		case 'find':
			if (isset($_POST['query']) && $_POST['query'] != '') {
				sendMpdCmd($sock, 'find ' . html_entity_decode($_POST['query']));
				echo json_encode(parseList(readMpdResp($sock)));
			}
			break;`

this is called by playerlib.js with:
`  else if ($(this).data('cmd') == 'track_info_pl') {
      if ($('#pl-' + (UI.dbEntry[0] + 1) + ' .pll2').html().substr(0, 2) != '<i') { // Has icon (fa-microphone)
        var title = $('#pl-' + (UI.dbEntry[0] + 1) + ' .pll1').html().trim();
        var artist = $('#pl-' + (UI.dbEntry[0] + 1) + ' .pll2').text(); // Artist - album
        $.post('command/moode.php?cmd=find', {'query': 'artist "' + artist  + '" Title "' + title  + '"'}, function(data) {audioinfo('track_info', data[0]['file']);}, 'json');
      }
  }`

2. Browse to folder from playlist-item or playbar-menu
Jump to the corresponding folder containing the playlist-song in folder view
(This is really a relief for me, as I'm still stuck in the "folder prison" as Tim once said :D )

For playlist-items this also uses the find extension in moode.php
called by playerlib.js with:
`  else if ($(this).data('cmd') == 'pl_browse_to_folder') {
    var filepath="";
    if ($('#pl-' + (UI.dbEntry[0] + 1) + ' .pll2').html().substr(0, 2) != '<i') { // Has icon (fa-microphone)
    	var title = $('#pl-' + (UI.dbEntry[0] + 1) + ' .pll1').html().trim();
    	var artist = $('#pl-' + (UI.dbEntry[0] + 1) + ' .pll2').text(); // Artist - album
    	$.post('command/moode.php?cmd=find', {'query': 'artist "' + artist  + '" Title "' + title  + '"'}, function(data) {
        filepath = data[0]['file'];
      	if (filepath!="") {
      		var dirDepth = 0;
      		filepath = filepath.slice(0,filepath.lastIndexOf("/"));
      		dirDepth = filepath.split("/").length - 1;
      		UI.dbPos[10] = dirDepth;
      		UI.dbEntry[3]='db-1';
      		mpdDbCmd('lsinfo', filepath);
      	}
      }, 'json');

    	currentView = 'playback,folder';
    	$('#coverart-url').click();
    }
  }`

for the current playing song it's easier:
`  else if ($(this).data('cmd') == 'track_browse_to_folder') {
    var filepath=MPD.json['file'];
    if (filepath && MPD.json['artist'] != 'Radio station') {
  		var dirDepth = 0;
  		filepath = filepath.slice(0,filepath.lastIndexOf("/"));
  		dirDepth = filepath.split("/").length - 1;
  		UI.dbPos[10] = dirDepth;
  		UI.dbEntry[3]='db-1';
  		mpdDbCmd('lsinfo', filepath);
    	currentView = 'playback,folder';
    	$('#coverart-url').click();
    }
  }`
  
3. multiple album-art
well - this is maybe interesting only for a few - like Desktop users, who mainly play local files.
I searched a bit and found that the ZEND lib only provides single pictures from the metadata - it wouldn't be much effort to get the whole array but they didn't implement that.
And after a quick search it seems to me, that there's no other usable solution for alac files....?
So for flac I'm using "metaflac" which is quite easy to use and to get the images from the file.
The next days I will take a deeper look into [getID3](https://github.com/JamesHeinrich/getID3) for other file formats.
At moment I got this far:
[image]

Clicking on the thumbnails on the right will open a modular with a full sized image 

Maybe some of these ideas are of interest

Cheers, Stephan

moodeaudio, 2021-01-18:
What are you are trying to accomplish with this issue?

Stephanowicz, 2021-01-18:
> What are you are trying to accomplish with this issue?

You mean the whole post?
It's just a suggestion for little enhancements which I hope are useful to the community. 
And I tried to show that it is not much effort to implement them. (the first two of them)
I dunno if the code fits to Your workflow and demands -  it's a suggestion how it can be done.

Cheers, Stephan

moodeaudio, 2021-01-18:
The challenge with converting from Zend to another metablock reader is that back when moOde 5 was released I had modified some routines in Zend to accept a "hash_only" param which when true results in an md5 hash of the first 1K bytes of the image to be returned. This is used in playerlib.js MPD.json['cover_art_hash'] to prevent unnecessary image reloads when playing an album. This would have to be carried forward in a replacement metablock reader.

Multi-art support including artist pics has been requested a few times but it can get complicated because sometimes the art is in the files and sometimes its a set of images in the album folder. I think image set is the more common way.

The thumbnail idea is cool. We could prolly do some neat things with that.

Stephanowicz, 2021-01-18:
Ah, ok - so if You already modified it, we could also enhance it with support for image arrays...? What do You think?

Regarding the other points - I'm a bit lazy and also annoyed of git regarding on how I can add new PRs in parallel. 
So I first wanted to check out if this is of interest before setting up a PR for this...

moodeaudio, 2021-01-18:
Zend only handles embedded images so even if it and other moOde code were modified to handle image arrays there would still need to be more modifications to code to handle multiple image files in the album dir.

Stephanowicz, 2021-01-20:
I think I've been dreaming of Zend being able to extract more than one pic - I can't find it anymore xD
Maybe I confused it with getID3...

Seems like zend is no longer under development? So in the longterm a change may be necessary

But anyways I'll checkout the possibilities of getID3 regarding imagearrays and will report here

Stephanowicz, 2021-01-20:
So - what do You think of my first 2 points - Trackinfo for playlistitems and Browse to folder for playlistitems? Is it of interest?

closed 2021-01-23 by Stephanowicz