moode-player/moode #313: Fresh install of raspios 2021-01-11 on RPi4 doesn't properly 'send' bluetooth audio to moode, or rather moode doesn't receive?

pickert, 2021-01-21:
Pairing and sound output selection works fine on the RPi4 desktop to connect to moode and moode web dashboard becomes greyed out as normal/BlueZ and shows a raspberry pi as the Bluetooth Active device... 

No sound comes through wired speakers though.. Other paired devices playbay fine.. Tested against Win10 desktop side by side to ensure moode/bluez were still stable and sound output ithrough speakers comes right back. 

Tested BT headphones against the RPI running the Raspberry PI desktop OS and its playback is fine also.. 

I'm guessing ''absolute'' volume control negotion/behavior might be the culprit via the newest changes to ALSA/PulseAudio in the 2021-01-11 raspbian build but can somebody test on the newest build also? Happy to run/provide any debugging/logging info that might help? But clearly it is an issue with bluealsa on the moode side from what I can tell?

dmesg shows the device when it pairs-
[  226.095736] input: C8:E2:65:9F:6B:C6 as /devices/virtual/input/input0

but in syslog I get just repeating -
``Jan 20 17:53:11 moode bluealsa[1073]: /usr/bin/bluealsa: W: Unsupported RTP payload type: 1
Jan 20 17:53:12 moode bluealsa[1073]: /usr/bin/bluealsa: W: Unsupported RTP payload type: 1
Jan 20 17:53:13 moode bluealsa[1073]: /usr/bin/bluealsa: W: Unsupported RTP payload type: 1
``

moodeaudio, 2021-01-21:
The syslog errors suggest the issue is in the RaspiOS/Pulseaudio client.

pickert, 2021-01-21:
Thanks Tim,
Headed to the raspbian forums to see if I can get some bites.. Appreciate the time and hard work on moode man! Great stuff!

moodeaudio, 2021-01-23:
I'll reopen if new info

closed 2021-01-23 by moodeaudio