moode-player/moode #318: Various UI issues

nakkore, 2021-01-23:
Hi,
I'd like to notify the following UI issues which I've found in the latest MoOde 7.0.1 by using Firefox 84.0.2 (64 Bit) Ubuntu 20.04.

* in Preferences changing the font size does not have any effect: refreshing the page or rebooting doesn't fix it.
* texts don't fit properly in component frames resulting vertically truncated (see attached png) probably texts are not vertically centered but I feel they're too bug anyway for the height of frames.
[image]
* theme background is not applied to all screens for example the background in the attached png was supposed to be "nightshade"
* in Preferences the cancel and update buttons are not vertically aligned 
[image]

Keep up the good work.

Thank you

moodeaudio, 2021-01-23:
Font size, theme settings, background, etc don't apply to the Config screens.

I'm not seeing the text or alignment issues on my end using either Safari or Chrome. Maybe it's unique to Firefox or a particular version of Firefox?

closed 2021-01-23 by moodeaudio

nakkore, 2021-01-23:
I'm not seeing the font size change anywhere anyway not just in the config screen. Where is it supposed to change ? 
Text issues don't appear on Chrome indeed nonetheless I'd have a look at Firefox as well.

moodeaudio, 2021-01-23:
I don't really have the time to test all Browsers so I just limit my testing to Safari since I do all the dev on a Mac, and Chrome since chromium-browser is used for the Local UI option. There are a lot of other users and testers in our Forum that use Firefox though so you might want to post your issue there and see if anyone can repro it. http://moodeaudio.org/forum

nakkore, 2021-01-23:
ok no problem. Thank you.

gbh-uk, 2021-01-24:
The vertical misalignment of the buttons appears in Chrome, Firefox and Edge for Windows.  
It's miniscule but can be resolved in moode.css: 60, by adding .2em margin top and bottom:

.modal-footer .btn + .btn {
    margin-left: 0;
    margin-bottom: .2em;
    margin-top: .2em;
}

The current rule  sets margin-bottom: 0; which overrides the margins originally set by the rule .modal-footer .btn in moode.css: 125.

moodeaudio, 2021-01-24:
Got it. I'll push a commit.