moode-player/moode #319: Configuring Wifi connections not possible in GUI

Rjay12, 2021-01-24:
I connected and set up Moode via Ethernet cable and everything is seems to be working including accessing network libraries. Unfortunately the WiFi connection seems broken. I have entered correct WiFi details in Network Config including my country and saved. Rebooted and removed my Ethernet cable. 

Moode does not connect over WiFi. Moode in not present on my network. I have no idea what is broken and I have not had such and issue with similar software.

Pi 3 B+
Release: 7.0.1 2020-12-23

moodeaudio, 2021-01-24:
You best bet is to post in the Support section of the moOde Forum http://moodeaudio.org/forum. Include the Moode startup log in your post. You can print it using SSH command 

```
moodeutl -l
```

Rjay12, 2021-01-24:
Thank you for getting back to me. 

I did try using the AP mode as well but WiFi is still broken. I don't expect support here - I'm just letting you know that there is an issue with WiFi in Moode since it doesn't work properly. It's free software so I don't expect anything from you. I'll just go back to Volumio and try out Moode again in the future to see if a working version is released.

Regards

moodeaudio, 2021-01-24:
You post look pretty trollish then. WiFi works for the vast majority of moOde users.

closed 2021-01-24 by moodeaudio

KicinskiJodlowski, 2023-10-19:
I have the same issue

KicinskiJodlowski, 2023-10-19:
[image]

KicinskiJodlowski, 2023-10-19:
[image]

photocurio, 2024-04-22:
I have the exact same console error as [KicinskiJodlowski](https://github.com/KicinskiJodlowski).