moode-player/moode #320: mpc outputdevice gets lost during reboots and defaults to CamillaDSP after reboot

Stephanowicz, 2021-01-26:
Hi,

this suddenly happend a few days ago -
I have an external USB-DAC. After start or reboot, the default device suddenly is CamillaDSP though I've never activated that...
Sometimes, when I forgot to switch the DAC on I got the appropiate error message - sometimes (?) it switched to the headphones output. There was no problem to switch on the DAC then and maybe restart pi or just mpd.

I realized that there was a minor update to 7.0.0.1 and after installing it seemed to work properly for  - I dunno 2-3 starts maybe.
The strange thing is, that in the UI everything is setup right - but even after saving, resetting etc. the CamillaDSP is active and no way to get the USB-DAC running.

I have to do that manually on commandline with mpc enable / disable...

**Output of mpc outputs after boot:**

> pi@moode7:~ $ mpc outputs
> Output 1 (ALSA default) is disabled
>         allowed_formats=""
>         dop="0"
> Output 2 (ALSA crossfeed) is disabled
>         allowed_formats=""
>         dop="0"
> Output 3 (ALSA parametric eq) is disabled
>         allowed_formats=""
>         dop="0"
> Output 4 (ALSA graphic eq) is disabled
>         allowed_formats=""
>         dop="0"
> Output 5 (ALSA polarity inversion) is disabled
>         allowed_formats=""
>         dop="0"
> Output 6 (ALSA bluetooth) is disabled
>         allowed_formats=""
>         dop="0"
> Output 7 (HTTP stream) is disabled
> Output 8 (ALSA CamillaDSP) is enabled
>         allowed_formats=""
>         dop="0"

the only message in **mpd/log** 
`Jan 26 10:58 : exception: Failed to read mixer for 'ALSA CamillaDSP': no such mixer control: PCM`

**moode.log**

> 20210125 231253 worker: -- Start
> 20210125 231253 worker: Successfully daemonized
> 20210125 231254 worker: Integrity check (warning: worker.php)
> 20210125 231254 worker: Integrity check (warning: playerlib.php)
> 20210125 231254 worker: Integrity check (warning: lib.min.js)
> 20210125 231254 worker: Integrity check (warning: main.min.js)
> 20210125 231254 worker: Integrity check (passed with warnings)
> 20210125 231254 worker: File check (OK)
> 20210125 231254 worker: Session loaded
> 20210125 231254 worker: Debug logging (OFF)
> 20210125 231254 worker: -- Device
> 20210125 231254 worker: ALSA cards: (0:b1 | 1:Headphones | 2:C1 | 3:empty
> 20210125 231254 worker: Configured: (2:Cambridge AudioDAC100 USB 1 | mixer:(PCM) | alsavol:none)
> 20210125 231254 worker: ALSA mixer actual (PCM)
> 20210125 231254 worker: ALSA PCM volume set to (amixer: Unable to find simple control 'PCM',0)
> 20210125 231254 worker: -- System
> 20210125 231255 worker: Host     (moode7)
> 20210125 231255 worker: moOde    (7.1.0 2021-MM-DD)
> 20210125 231255 worker: RaspiOS  (10.6)
> 20210125 231255 worker: Kernel   (5.4.77-v7l+ #1371)
> 20210125 231255 worker: Platform (Pi-  )
> 20210125 231255 worker: ARM arch (armv7l, 32-bit kernel)
> 20210125 231255 worker: MPD ver  (0.22.3_p0x3)
> 20210125 231255 worker: CPU gov  (ondemand)
> 20210125 231255 worker: USB boot not available
> 20210125 231255 worker: File system expanded
> 20210125 231255 worker: HDMI port on
> 20210125 231255 worker: -- Network
> 20210125 231255 worker: eth0 exists
> 20210125 231255 worker: eth0 wait 0 for IP address
> 20210125 231258 worker: eth0 wait 1 for IP address
> 20210125 231301 worker: eth0 wait 2 for IP address
> 20210125 231304 worker: eth0 wait 3 for IP address
> 20210125 231307 worker: eth0 wait 4 for IP address
> 20210126 105759 worker: eth0 address not assigned
> 20210126 105759 worker: wlan0 exists
> 20210126 105759 worker: wifi country (DE)
> 20210126 105759 worker: wlan0 trying SSID (JESUS ROCKS!)
> 20210126 105759 worker: IP addr (192.168.178.28)
> 20210126 105759 worker: Netmask (255.255.255.0)
> 20210126 105759 worker: Gateway (192.168.178.1)
> 20210126 105759 worker: Pri DNS (192.168.178.1)
> 20210126 105759 worker: Domain  (fritz.box)
> 20210126 105759 worker: -- Audio
> 20210126 105800 worker: MPD conf updated
> 20210126 105800 worker: ALSA outputs unmuted
> 20210126 105800 worker: ALSA card number (2)
> 20210126 105800 worker: Audio output (Cambridge AudioDAC100 USB 1)
> 20210126 105800 worker: Audio formats (S24_3LE)
> 20210126 105800 worker: ALSA mixer name (PCM)
> 20210126 105800 worker: MPD volume control (software)
> 20210126 105800 worker: Hdwr volume controller not detected
> 20210126 105800 worker: Reset renderer active flags
> 20210126 105800 worker: -- MPD
> 20210126 105800 worker: MPD started
> 20210126 105800 worker: MPD accepting connections
> 20210126 105800 worker: Configure MPD outputs
> 20210126 105800 worker: MPD output 1 ALSA default (off)
> 20210126 105800 worker: MPD output 2 ALSA crossfeed (off)
> 20210126 105800 worker: MPD output 3 ALSA parametric eq (off)
> 20210126 105800 worker: MPD output 4 ALSA graphic eq (off)
> 20210126 105800 worker: MPD output 5 ALSA polarity inversion (off)
> 20210126 105800 worker: MPD output 6 ALSA bluetooth (off)
> 20210126 105800 worker: MPD output 7 HTTP stream (off)
> 20210126 105800 worker: MPD crossfade (off)
> 20210126 105800 worker: -- Feature availability
> 20210126 105800 worker: Source select (available)
> 20210126 105800 worker: Source select (source: MPD)
> 20210126 105801 worker: Source select (output: Cambridge AudioDAC100 USB 1)
> 20210126 105801 worker: Bluetooth (available)
> 20210126 105801 worker: Airplay renderer (available)
> 20210126 105801 worker: Spotify renderer (available)
> 20210126 105801 worker: Squeezelite renderer (available)
> 20210126 105801 worker: UPnP renderer (available)
> 20210126 105801 worker: UPnP renderer (started)
> 20210126 105801 worker: DLNA server (available)
> 20210126 105801 worker: UPnP browser (available)
> 20210126 105801 worker: GPIO button handler (available)
> 20210126 105801 worker: Stream recorder (n/a)
> 20210126 105801 worker: -- Music sources
> 20210126 105801 worker: USB source (MEDIA1)
> 20210126 105801 worker: NAS and UPnP sources (none configured)
> 20210126 105801 worker: -- Miscellaneous
> 20210126 105801 worker: USB volume knob (Off)
> 20210126 105801 worker: USB auto-mounter (udisks-glue)
> 20210126 105801 worker: LED0 (On)
> 20210126 105801 worker: LED1 (Off)
> 20210126 105801 worker: Saved MPD vol level (0)
> 20210126 105801 worker: Preamp volume level (0)
> 20210126 105801 worker: MPD volume level (20) restored
> 20210126 105801 worker: ALSA volume level (None)
> 20210126 105801 worker: Auto-play (Off)
> 20210126 105801 worker: Maintenance interval (3 hrs)
> 20210126 105801 worker: Screen saver activation (60)
> 20210126 105801 worker: Session permissions (OK)
> 20210126 105801 worker: Watchdog started
> 20210126 105801 worker: Ready
> 20210126 105901 sendEngCmd(): File open failed, UI has never been started
> 

I hope smbd knows what's going on here.

Cheers, Stephan

moodeaudio, 2021-01-26:
The log is missing the CamillaDSP line which suggests not all the files in your build are current with the repo.

```
.
.
20210125 221155 worker: Reset renderer active flags
20210125 221155 worker: CamillaDSP (off)
20210125 221155 worker: -- MPD
20210125 221155 worker: MPD started
20210125 221156 worker: MPD accepting connections
.
.
```

Stephanowicz, 2021-01-26:
Thank You! *thumbs up*

Copied the worker.php as it was missing the whole camilla stuff... (I wonder why?) - restarted and settings where properly restored.
Hope it lasts :D - sometimes I wonder what's going on in there...

Cheers, Stephan

closed 2021-01-26 by Stephanowicz

Stephanowicz, 2021-01-26:
Ok - I was too early :D 
The worker didn't start because the new camilla files were missing....
Seems like shouldn't update from the gui anymore when having files from development in my build... :( 
So the problem is still the same - with the log having the line You mentioned before. Maybe there are database entries missing...

Ok, either I will live with that or I will have to update all my files with my develop-branch.