moode-player/moode #329: forum ban

MatteoEng, 2021-02-02:
Hi, I subscribed to moode forum a couple of ours ago, I made a post where I asked help for the GUI, then my entire network got a permaban...the message said to contact an admin but it doesn't let me do it...so sorry if I write here but I don't have any other place to do.
First of all, what did happen?
why am I banned?
how can I ask for help?
it's really weird

moodeaudio, 2021-02-02:
it was prolly flagged as either SPAM or Commercial. I've re-activated your account.

closed 2021-02-02 by moodeaudio

MatteoEng, 2021-02-03:
for now i am still banned... the entire network is banned...may i have any updates? thanks

moodeaudio, 2021-02-03:
Very odd that the address was still in the ban list after lifting the ban. Its supposed to be deleted automatically. In any case, I've manually removed it so you should be able to access the Forum now.

MatteoEng, 2021-02-03:
thanks so much, I really appreciate