moode-player/moode #331: Roon lock audio output

MatteoEng, 2021-02-05:
Hi, I installed the roon bridge that I found on your website, it works really good, but then when I try to switch to another source, like airplay, bluetooth... I have no sound in output, then if I resume roon, the sound came out, any suggest? maybe how to swtich from a source to another 
P.S.
even if I close the roon server it continually tries to play through roon
thanks
-Matteo

moodeaudio, 2021-02-05:
Theres no support in moOde for managing roon bridge as a renderer but even if support were added, rb prolly acts like squeezelight which defaults to hogging the audio output. Theres an option flag for sl that causes it to release the audio output after N secs but it has to be added to the "Other options" field in Squeezelite Config. I don't know whether rb has such an option.

The other issue you will run into is that there is no provision in moOde for one renderer interrupting another. It may be possible to implement a mechanism to do this but generally users connect via one renderer of choice and rarely switch to another. This makes it an edge case and so won't get much if any priority.

MatteoEng, 2021-02-05:
Ok, thanks so much, in case I will update you if I find a way to do it.
-Matteo

closed 2021-02-07 by moodeaudio