moode-player/moode #333: Feature request: Simple HiRes and S/PDIF support

mend0za, 2021-02-08:
Yesterday I installed moOde on my RasPI 3 to use it as Audio Receiver and recognize that some tweaks for HiRes are still missing.

Could you please add switches for  S/PDIF output and possibility to specify output format?

As I using FiiO K3 USB DAC and prefer to output sound via  optical S/PDIF, so I made some study and  fast and simple hacks for /etc/mpd.conf generator:
```php
index cf38200c..c6c92615 100644
--- a/www/inc/playerlib.php
+++ b/www/inc/playerlib.php
@@ -2000,7 +2000,7 @@ function updMpdConf($i2sdevice) {
 
        // ALSA local (outputs 1 - 5)
        $names = array (
-               "name \"ALSA default\"\n" . "device \"hw:" . $device . ",0\"\n",
+               "name \"ALSA default\"\n" . "device \"iec958:" . $device . ",0\"\n",
                "name \"ALSA crossfeed\"\n" . "device \"crossfeed\"\n",
                "name \"ALSA parametric eq\"\n" . "device \"eqfa12p\"\n",
                "name \"ALSA graphic eq\"\n" . "device \"alsaequal\"\n",
@@ -2013,6 +2013,11 @@ function updMpdConf($i2sdevice) {
                $data .= "mixer_type \"" . $mixertype . "\"\n";
                $data .= $mixertype == 'hardware' ? "mixer_control \"" . $hwmixer . "\"\n" . "mixer_device \"hw:" . $device . "\"\n" . "mixer_index \"0\"\n" : '';
                $data .= "dop \"" . $dop . "\"\n";
+               if (preg_match("/ALSA default/", $name))
+               {
+                       $data .= "format \"96000:32:2\"\n";
+                       $data .= "auto-resample \"no\"\n";
+               }
                $data .= "}\n\n";
        }
```
I forced MPD to 96KHz/32 just because of limitation of my multimedia speakers (its toslink not a HiEnd :) ) . 

moOdy UI works like a charm  and detect advanced sound settings properly : 
[image]

on ALSA low level everything is also good 
[proc_asound_K3_stream0.txt](https://github.com/moode-player/moode/files/5943454/proc_asound_K3_stream0.txt)

I'm not familiar with moOdy code, so I still don't know where to add it to see via the http://ip/mpd-config.php

moodeaudio, 2021-02-08:
These hacks aren't needed because MPD/SoX performs the same function and its already available in the MPD Config screen. Also your audio device already supports 96K rate and so your 96K FLAC file will be played at that rate as-is without any conversion needed.

mend0za, 2021-02-08:
My active speakers didn't play 24/192Khz via S/PDIF, the highest  reliable combination of DAC + speakers I found is 32bit/96Khz.

If I allow to do automatically settings select for MPD/SoX some of my hires collection will play silence (typical 24 bits/192Khz for example). K3 dac will play them correctly without re-sampling on analog and digital output, but speakers amp will not understand the stream via s/pdif.

So it's a reason why I force the alsa output settings to  32bit/96Khz explicitly. And SoX re-sampling is turned off in my configuration.

And what about setting for using S/PDIF digital output (device "iec958:2,0") instead of analog one  (device "hw:2,0") ? In several hardware configurations it will significantly decrease noise and unnecessary clicks.

moodeaudio, 2021-02-08:
You are still using SoX resampling in your hacks. The "format" line does same thing as "audio_output_format" line used by moOde SoX config. The difference is that "format" is per-output while "audio_output_format" is global and applies to all outputs. https://www.musicpd.org/doc/html/user.html#global-audio-format

```
$data .= "format \"96000:32:2\"\n";
```

The hack below does not turn off SoX resampling it only turns off the low quality libasound resampler which isn't going to be used anyway. https://www.musicpd.org/doc/html/plugins.html#alsa-plugin The param is also misspelled in your example and thus is not having any effect. It should be "auto_resample" with an underscore. 

```
$data .= "auto-resample \"no\"\n";
```

Below is an example of how the global SoX settings appear in mpd.conf.
.
.
filesystem_charset "UTF-8"
metadata_to_use "+comment"
**audio_output_format "96000:32:2"**
replaygain "off"
replaygain_preamp "0"
volume_normalization "no"
audio_buffer_size "4096"
max_output_buffer_size "131072"
max_playlist_length "16384"
max_connections "128"
.
.

I don't really understand the statement below because moOde is connected via USB, a digital connection, to your device. moOde only knows about the USB interface and not about the various line-level outputs that the device provides. Those are handled internally by the firmware in the device.

```
"And what about setting for using S/PDIF digital output (device "iec958:2,0") instead of analog one (device "hw:2,0") ? In several hardware configurations it will significantly decrease noise and unnecessary clicks."
```

mend0za, 2021-02-08:
Thank you for patience and explanation!

Yes "auto_resample" is correct with underscore, I fixed it locally later, but not in initial post, sorry.

Now I finally understand that SoX always do re-sampling in mpd context for "audio_output_format" and "output_format". 

Thank you for your great work! 
I dropped my changes and reconfigure re-sampling via SoX. It made the same effect at the end.

closed 2021-02-08 by mend0za