moode-player/moode #339: making stationmanager.py a first-class citizen

TheOldPresbyope, 2021-02-19:
Currently, exporting and importing stations is done with a minimum of user interaction. On import, just a few lines are written to the WebUI telling of success or failure.

Since the new script outputs a goodly amount of information, one might consider writing it to a /var/log/stationmanager.log file (perhaps only when moOde debugging is enabled?).

Right now the moode.log is silent about this task. One might also consider logging to it that the script ran and include the return code.

bitkeeper, 2021-02-20:
FYI:

No full of you want, but at least in case of issue during the import the full output of stationmanager is shown:
```php
	exec($cmd, $output, $exitcode);
					if($exitcode!=0) {
						$msg= implode("\n", $output);
						str_replace("\"", '', $msg);
					}
```

And as a firs-class citizen setting the filter options on import would be great.
I think most user only will need import for scope 'other' and get the updates from the moOde stations on each update.

TheOldPresbyope, 2021-02-20:
Ah, missed that $output is captured already. (I think my brain must be affected by the current cold snap. This isn't my first unforced error of the week.)

I agree that many probably just want to import 'other' stations although we've had a number of users who pared back the curated list by a lot and don't want to have to do it again on each major release. There's a conflict of interest here, of course, since station URLs change from time to time.

moodeaudio, 2021-03-15:
Kent, can this be closed?

TheOldPresbyope, 2021-03-15:
I'll close it for now and wait to see what the user experiences are.

closed 2021-03-15 by TheOldPresbyope