moode-player/moode #354: "mpc volume" changes are not updated in web ui

etgocode, 2021-02-27:
If you set the volume with ssh like `:~$ mpc volume 30` the web ui will not show the volume change. Refreshing, reloading or closing and restarting the browser does not help.

moodeaudio, 2021-02-27:
Try 
```
/var/www/vol.sh --help
```

closed 2021-02-27 by moodeaudio

etgocode, 2021-02-27:
> /var/www/vol.sh --help

I don't see how this is helping.
The volume knob works, it just does not show the correct value after the volume has been set via the mpc command. Changing the volume on the knob afterwards correctly sets it to the value the knob is displaying. But you don't know the volume level _before_ you set it with the knob.

moodeaudio, 2021-02-27:
vol.sh is the mechanism by which external applications can control moOde volume and mute state including updating the knob. Just using mpc won't work.

etgocode, 2021-02-27:
Awesome! `vol.sh 30` does exactly what I need. Unfortunately the help text didn't tell that you can set a specific value, that's why I didn't get what you meant the first time.
Thanks for your support and keep up the good work!😊

moodeaudio, 2021-02-27:
It's definitely explained in the help but if you think its not clear enough then suggest improved wording.

pi@rp2:~ $ /var/www/vol.sh --help
Usage: vol.sh [OPTION] [VOLUME]
Change the volume and update the knob.

With no OPTION or VOLUME, print the current volume.
**With just VOLUME, set current volume to VOLUME**

 -up            VOLUME  value between 1 and 100
 -dn            VOLUME  value between 1 and 100
 -mute          mute or unmute the volume
 -restore       set volume to current knob level
 --version      print the program version
 --help         print this help text

etgocode, 2021-02-27:
Oh, I get it know, my bad 😓
Maybe it could be changed to 
```
Usage: vol.sh [OPTION] [VOLUME]
Change the volume and update the knob.

With no [OPTION] or [VOLUME], print the current volume.
With just [VOLUME], set current volume to [VOLUME]

Options are:

-up [VOLUME] value between 1 and 100
-dn [VOLUME] value between 1 and 100
-mute mute or unmute the volume
-restore set volume to current knob level
--version print the program version
--help print this help text
```

for easier readability.