moode-player/moode #366: Add HiFiBerry DAC2 HD

gregvds, 2021-03-08:


closed 2021-03-08 by moodeaudio

HolyMacarony, 2022-10-31:
Hello everyone, i just got an Hifiberry DAC2 HD + DSP ADD-ON BOARD installed and runing Moode Audio. 
In moodeaudios audio Settings tab under "Named I2S device" there is no entry like "Hifiberry DAC2 HD" or even "Hifiberry DAC2 HD + DSP". 
The only thing referring the Hifiberry DAC2 HD is a dtoverlay=hifiberry-dacplushd, since this is also the only info, the hifiberry website gives about using this DAC. 
So far so good. The DAC is working with this dtoverlay option. but what do i have to do in order to make use of the DSP module in combination with the DAC?
I could not find anything on this regard anywhere and also the hifiberry support did not want to give support for this. So iam kindly asking whether anyone has a clue on how this should/could be working? 
Because when i install the DSP Module i get every so often a strange buzzing noise at the beginning of songs for like 10 secounds. It has to have to do with the DSP not wokring correctly (and yes i tested the DSP unter Hifiberry OS, it works an gets recognised. Its just the Hifiberry OS is completely unuseable in any other regard beside detectiong the hardware installed...)
i would very much appreciate any kind of thoughts about this topic. Thank you all for this great software!

moodeaudio, 2022-10-31:
I have no idea how those devices work but It sounds like they require the vendors custom OS.