moode-player/moode #381: Suggestion: display a simple bar EQ on the 'x Active' overlay screen

ahydronous, 2021-03-28:
Same idea as my other suggestion, I feel like it would it a little visual flourish to an otherwise spartan screen.
I'm not sure if its called a 'bar EQ' officially, I mean the 8+ bars that go up and down based on the music.
It could even be tinted to the source, so Spotify green for Spotify, white for Airplay and turquoise for Squeeze.
In this case, a good location might be along the bottom of the screen?

moodeaudio, 2021-03-28:
This might be doable. We already use an EQ SVG for the now-playing icon. I'll add to the TODO list to investigate.

gbh-uk, 2021-03-28:
If it's implemented please also consider an option to turn it off.  There was some push-back from users when the animated 'now-playing' icon was introduced.  Personally I'd like to see an option to replace that with a static one too.
I started to code such an option but got side-tracked by work so never got to a full PR.

moodeaudio, 2021-03-29:
Yes, I remember that. There is a TODO list item to have choices for the now-playing icon but it never got priority, at least for me to work on it.

I did do a bit of investigation and mocked up a renderer overlay with the EQ icon and it was very distracting to my eye. The other issue is that it would constantly be active unless there were code to manage its state and turn it off when the renderer was not playing.

moodeaudio, 2021-04-24:
Closed due to lack of activity

closed 2021-04-24 by moodeaudio