moode-player/moode #385: web UI becomes unresponsive when connecting multiple times from the same client

slybit, 2021-04-02:
**Experience behavior:**

I noticed that the web UI becomes unresponsive when connecting 2 or more times from the same client machine. For example when opening 2 tabs in my browser, both pointing to the Moode audio web UI, they both become unresponsive. The moment I remove 1 of the 2 tabs, the other becomes responsive again.

**Expected behavior:**

Having 2 tabs open that both can interact with the Moode audio web UI.

**Notes:**

- This is using Moode audio version 7.1.0 running on a Raspberry pi 3. Nothing else is running on that machine.
- This is more of a nuisance than an issue. In my case, I also run the UI in a frame in the Home Assistant UI, so the moment both the Home Assistant UI and the normal Moode audio UI are opened in my browser, the freeze occurs. Perhaps other users have similar use cases where this is an issue.

moodeaudio, 2021-04-02:
There are no complaints in the Forum but I suspect thats because it's an uncommon usage scenario to have multiple tabs or windows open on the same client to the same moOde player.

I am able to partially repro though but it has to be > 2 tabs or windows open on the same Browser on the client. Opening a 3rd tab or window causes very slow or unresponsive WebUI on all tabs/windows. 

What works perfectly is the following where there are 6 instances of the WebUI running.

iMac
- 2 tabs open on Safari
- 2 tabs open on Chrome

iPad
- 1 Safari window open

iPhone
- 1 Safari window open

This would suggest that the issue is not related to capacity or performance of the server when handling multiple client connections but maybe something to do with the Browser or with the way the code behind the WebUI is implemented. I don't have enough expertise in Browsers to be able to debug.

monkeypox, 2021-04-03:
I looked at this a little and it’s not so much that the webui is unresponsive but the “server” doesn’t seem to be responding. Eventually it’ll load content but it’s very slow until you kill the third instance (macOS chrome). 

This could be a shared browser bug as moode is definitely working with no issues when accessed through another client (e.g. my phone) while still flailing on my laptop but it could also be some weird nginx edge case. Unfortunately I can’t help with either of those things but will continue to look around a bit to see if this has come up in other situations and if there were any solutions offered.

moodeaudio, 2021-04-03:
It could be server-side issue that unrelated to perf or capacity. Maybe some sort of blocking is occurring, possibly in sqlite or session updating. It's odd that its only when > 2 tabs ...

monkeypox, 2021-04-03:
One other data point: two mobile safari instances shows the issue (though it’ll load the second) but one mobile safari plus one added to homescreen instance doesn’t. 

Maybe some kind of default ddos type protection with too many connections from the same ip? It isn’t SQLite because it won’t even load images.

moodeaudio, 2021-04-03:
Out of curiosity, try looking at the Moode log for errors from watchdog. I'll also check next time I try to repro this.

monkeypox, 2021-04-03:
Nothing in moode.log but syslog has this (just the fastcgi stuff at first then I closed and loaded a new instance and scrolled the album list)...

`Apr  3 19:33:07 moode systemd[1]: Stopping The PHP 7.3 FastCGI Process Manager...
Apr  3 19:33:07 moode systemd[1]: php7.3-fpm.service: Succeeded.
Apr  3 19:33:07 moode systemd[1]: Stopped The PHP 7.3 FastCGI Process Manager.
Apr  3 19:33:07 moode systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Apr  3 19:33:07 moode systemd[1]: Started The PHP 7.3 FastCGI Process Manager.
Apr  3 19:35:01 moode rngd[369]: stats: bits received from HRNG source: 1680064
Apr  3 19:35:01 moode rngd[369]: stats: bits sent to kernel pool: 1623072
Apr  3 19:35:01 moode rngd[369]: stats: entropy added to kernel pool: 1623072
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2 successes: 84
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2 failures: 0
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2(2001-10-10) Poker: 0
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2(2001-10-10) Runs: 0
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2(2001-10-10) Long run: 0
Apr  3 19:35:01 moode rngd[369]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Apr  3 19:35:01 moode rngd[369]: stats: HRNG source speed: (min=277.141; avg=518.057; max=536.883)Kibits/s
Apr  3 19:35:01 moode rngd[369]: stats: FIPS tests speed: (min=3.630; avg=17.311; max=31.423)Mibits/s
Apr  3 19:35:01 moode rngd[369]: stats: Lowest ready-buffers level: 2
Apr  3 19:35:01 moode rngd[369]: stats: Entropy starvations: 0
Apr  3 19:35:01 moode rngd[369]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
`

moodeaudio, 2021-04-04:
I always see the entropy stuff in syslog but the "stop / start FastCGI" entries are suspicious.

slybit, 2021-04-04:
I can confirm that the issue occurs when opening the 3th tab. Also multiple, different clients are not problem as stated above.

I enabled the php-fpm logs, but nothing in there except
```
[04-Apr-2021 05:12:21] WARNING: [pool www] child 7816 said into stderr: "amixer: Unable to find simple control 'none',0"
[04-Apr-2021 05:12:21] WARNING: [pool www] child 7816 said into stderr: ""
```
These 2 lines occur every time the Mo0de UI is refreshed, _independent of the number of open tabs_. So I assume it is not related.

moodeaudio, 2021-10-21:
Lets reopen if there is new information on the issue.

wwigman, 2022-08-20:
I can conform ui unresponsive.
Pi 4B 4GB streaming from synology nas, RS815.
Pressing next song button, after 10 seconde the next song, 10 seconde later the screen is refreshed.
Library consists of 3500 FLAC and WAv songs. 24bit 44 kHz ( i converted al my CD and LP's to digital)

moodeaudio, 2022-08-23:
Thats a different issue. We handle these types of issues in our Support Forum. Email tim at moodeaudio dot org with your preferred userid and I'll create an account for you.

jeanrocco, 2023-04-22:
Hello

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro9,1
  Processor Name:	Quad-Core Intel Core i7
  Processor Speed:	2.6 GHz
  Mac OS: Ventura 13.3.1 using https://dortania.github.io/OpenCore-Legacy-Patcher,

WIth Firefox 112.0.1 I am experiencing this sluggish response from Moode 8.3.1's web UI, when using http://moode as URL, but if I connect first to Moode with " ssh -L8080:localhost:80 pi@moode.ip.address " and then use " localhost:8080 " as the URL the UI works flawlessly, and is very fast. I have experienced sluggish response like that with other Web UI from old cameras and even from Home Assistant. The " ssh -L ... " trick always solve the problem. I can only guess the implementation of Firefox on the Mac is causing that. Any thoughts ?

And btw, thanks for this great software

jrb.

moodeaudio, 2023-04-22:
IME slow WebUI is almost always due to some sort of network or Browser issue external to moOde software. 

You could try Chrome or Safari Browser without any extensions loaded, or switch to Ethernet connected directly to your Router and see if either of these makes a difference.

closed 2023-04-22 by moodeaudio

jeanrocco, 2023-04-22:
thank you for your answer , bye, jrb.

jeanrocco, 2023-04-22:
Hello Tim, a final word before closing time :-) ! The web UI works very
well when accessed from my Linux PC (Linux Mint 20 Cinnamon) with Firefox
(mint-001) so apparently the problem lies with Firefox running on a MAC ...
I just hope I didn't make you lose too much of your time, just wanted to
let you know ! bye . jrb.