moode-player/moode #401: can't set path for smb share

brainchild0, 2021-05-15:
Currently, samba shares must be given by host name and share name only, without the ability to restrict the path within the share. Doing so is often valuable for a variety of reasons.

A few possible overall enhancements to samba sharing might include:

- Allow user to specify specific path in samba share.
- Allow user to browse file tree in samba to choose such path.
- Allow user to choose from samba server detected by uPnP.

moodeaudio, 2021-05-15:
After selecting server/share from the SCAN dropdown list click "EDIT path manually" You can replace host name by IP address and also add on a directory path after the share name.

The SCAN dropdown list is generated from the command below.

```
smbtree -N -b
```

What command is used to list Samba servers via UPnP?

brainchild0, 2021-05-15:
I thought uPnP was the protocol used to scan for samba servers, but I may be wrong. The scan  button had no effect for me even though the share was reachable by manual entry.

At the moment, the message for *Manual Path Entry* is "Format is Host/Share or IPaddr/Share. Up to 64 chars, leading slash / not allowed". I interpreted this message to mean that only a share name, not a path within a share, is allowed. I'm not sure why the character limit exists on the path, but perhaps the message could be revised to more clearly indicate that a path with the share is allowed.

brainchild0, 2021-05-15:
Also, as an aside, if no user name is given for adding a new samba share, then the  interface returns with an error and discards the input from the form. Two changes might be considered. First, the interface might accept the empty input for a user name. Second, if the submitted data is for any reason not valid, then the form  might be restored with the previous   data populating it.

moodeaudio, 2021-05-16:
You will be better served by posting these kinds of questions in the support section of our Forum.
http://moodeaudio.org/forum

closed 2021-05-16 by moodeaudio

brainchild0, 2021-05-16:
If you're not actively using this tracker for general reports, you might add a template, with an explanatory note in a comment block to point potential submissions to the appropriate location.

moodeaudio, 2021-05-16:
Thats prolly a good idea although we typically only get issues from devs about the code or other such things.

How do I do a template and a comment block?

brainchild0, 2021-05-16:
I've never administrated  a public repository, but from current understanding of Github functionality, I would guess it is done through committing a hidden file, or directory, to your source root.

Most projects with a public user base take issues from the public through the issue tracker associated with the repository.   I wouldn't think to look for a different location unless prompted by some instructions.

brainchild0, 2021-05-16:
https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository

moodeaudio, 2021-05-16:
The problem with GitHub Issues is that it's just a flat list while Forums offer a rich, threaded discussion format which is really nice for providing suppose for users. 

Like I mentioned earlier Git issues in our project are mainly for devs and their questions about the code or the project.

brainchild0, 2021-05-16:
The suggestion I was making was to put a reference to your forum in the issue template, that a new participant would read before completing a submission to the tracker, if it is your preference to receive reports   in the forum. 

I agree that a forum has a nicer feel for support, but my submissions were feature requests and bug reports, not support inquiries.

moodeaudio, 2021-05-16:
The Forum provides for feature requests, bug reports which are "support" etc.

Whats your point in continuing this discussion? If you want to get support register in the Forum.

brainchild0, 2021-05-16:
I was only making a suggestion, in case someone else might follow the same path I had followed. Sorry for any confusion.