moode-player/moode #403: support for cue sheets and single-file disc rips

brainchild0, 2021-05-16:
The handling for cue sheets appears to be different from the handling by many other systems. 

Moode gives the user an option to ignore   cue sheets. If this option is not selected, then Moode considers the cue sheet as a library entry separate from any audio files.

Neither option appears to be optimal to a user with cue sheets in a music library.

Most players handle cue sheets as follows:

1. For each cue sheet, process the data  as though it were tag data in the audio files, and ignore the actual corresponding metadata in the audio files.
2. If the cue sheet refers to a single file for the entire disc, then present each track as a separate song in the library, as though each song were a separate file. During playback for the disc, play the file continuously, but display track data from the entry for the track  in the cue sheet.
3. If the cue sheet refers to a separate file for each disc, then during playback of the disc, play each file in sequence, attempting to mimic the timing of the original disc, described in the cue sheet.

I am wondering whether for any reason it would not be preferred for Moode to adopt similar behavior.

moodeaudio, 2021-05-16:
You will need to post these kinds of questions in the support section of our Forum.
http://moodeaudio.org/forum

closed 2021-05-16 by moodeaudio

mabushey, 2022-09-28:
I've been "trying" to use Raspberry Pi's as 16/14bit music players for over half a decade. Some of the music players have taken multiple days to index only a few thousand albums, and most are too slow to be usable once indexed. No one really seems to care about single file flacs with cuesheets. Maybe 5 years ago Rune audio worked perfect regarding cuesheets and usability speed, but they regressed. I may eventually release (I'm working on video indexing right now) a Rust based system than can replace Kodi and the Pi music players. The database would be optionally local, but preferably remote; these Pi's need everything they have just to run a UI and decode bits. Done right, databases can be very fast, unlike ALL the existing Pi music player systems.

moodeaudio, 2022-09-28:
Based on the troubleshooting in our Forum 99% of "Indexing is slow" is external to moOde software. The bottlenecks are typically network, NAS, slow HDD, slow boot media or hardware issue. Extreme sized collections containing 100's of thousands of tracks are also problematic.

Flac with non-embedded cue sheet works perfectly on my end and for many of our users :-) Here are some screen shots for the cue/flac in the following directory. Note that there is no support in moOde for flac with embedded cue sheets or any other formats with embedded cue.

```
pi@moode:~ $ ls -l /media/VFAT128/Test_mixed/cue-test-no-embedded/
total 229504
-rw-r--r-- 1 root root      1388 Jun 27 02:56 Spectrum.cue
-rw-r--r-- 1 root root 234972804 Jun 27 02:56 Spectrum.flac
```
[image]
[image]
[image]

@ljufa created a Rust based player for his DIY project. You might want to check it out.
https://github.com/ljufa/rsplayer

Btw,  good luck with you project.