moode-player/moode #415: Simplify donations by adding funding through Github

gruentee, 2021-06-30:
Hi, 

first of all – thanks for making moode. 

Second – when trying to donate I noticed there is no sponsorship button to be found at the Github page. Is there any reason against it? If not I'd suggest adding it in order to make it easier for people to support the project via donations.

Best
Constantin

moodeaudio, 2021-06-30:
Hi,

I didn't know that was available. I'll check it out, thanks :-)

moodeaudio, 2021-07-01:
After reviewing it, the GitHub sponsorship system seems a bit too involved for our project at this time. I did add a simple PayPal button to the repo though which I think will make it easier for people to donate.

closed 2021-07-01 by moodeaudio

gruentee, 2021-07-05:
Okay, seems reasonable.