moode-player/moode #426: add new driver and overlay for our I2S-cards

chipdipru, 2021-07-28:
Hello, we have prepared a driver and overlay for our I2S-cards. 
https://github.com/chipdipru/ChipDip-DAC-driver-Moode
How can we add it natively in Moode-player?
Which way?
Regards

moodeaudio, 2021-07-28:
Our project only supports I2S drivers that are included in the stock Raspberry Pi Linux kernel.
https://github.com/raspberrypi/linux

closed 2021-07-28 by moodeaudio

chipdipru, 2021-07-28:
Our driver is included in the standard kernel. https://github.com/raspberrypi/linux/pull/4476
We can expect inclusion in Moode-player?

moodeaudio, 2021-07-28:
It's in the 5.10 branch and our current release plan is to stay on RaspiOS 10.6 and kernel 5.4.x until we feel its necessary to bump to the latest RaspiOS and 5.10 kernel branch. I don't have a timeframe for that.