moode-player/moode #427: [Feature] Allo Boss 2 OLED display and remote integration

diist, 2021-07-28:
The Allo Boss 2 OLED display, controlled by a stick button or the remote, shows DAC settings such as volume, sampling info, and further options in the settings like phase compensation and high-pass filter. 

I half-assedly rewrote the `/opt/boss2_oled/boss2_oled.py` script that handles the display to instead show volume and current track and the remote to play / pause / volume +- / track +-; it does so by running `mpc` commands. But before I go further I was hoping to gauge interest and gather requirements. So far I have something like:
```
VOLUME: 60%
ARTIST_LONG_NAM
E - TRACK_LONG_N
AME_THAT_IS_TRUN
```
When playing `Artist_long_name - Track_long_name_that_is_truncated`

1. [ ] A) Given that we can set the DAC `Chip Options` in the UI already, do we care about keeping them accessible from the stick button or remote? So far I've got the stick still on the original behaviour and the remote issuing `mpc` commands, but I'd rather get rid of the original behaviour so that I can redo the screen font / pixel layout without worrying about the existing screens.
2. [ ] B) Should the new UI be the default behaviour when using a Boss 2? Or should we be able to toggle it on / off in the UI (I have zero front-end foo)
3. [ ] C) When do we put the screen to sleep? At minima we want it on while mpc is playing but what about the renderers? For example I don't know how to tell when spotify is playing anything, we could just show the volume for a while then sleep in that case. 
4. [ ] D) Do we care about seeing the sampling info? There's no room on the screen for volume _and_ track name _and_ sampling info
5. [ ] E) How do we show the track name given the screen limitations? Truncate it over multilines like I have done? It's pretty ugly. I could try doing a rotating type display but that might be flashy and weird.

moodeaudio, 2021-07-28:
The challenge in displaying song info on the Boss2 OLED is that the screen is really tiny. You might want to post the script in the Moode Forum to gauge interest. Personally I prefer just volume and format on such a small screen.

diist, 2021-09-06:
won't do

closed 2021-09-06 by diist