moode-player/moode #428: Simple gui for ALSA graphic eq

Stephanowicz, 2021-07-31:
Hello there!

I was a little bit annoyed with the usage of the graphics eq in moOde
e.g. when de-/activating the eq mpc is stopped
new values have to be activated - they are not instantly set

By searching for gui's for alsaeq I realized that this is not a necessary thing - You just need to pause mpc , switch the output and then can instantly start playback again.
Also alsamix is capable to change the EQ settings on the fly...

I finally found a nice gui from [elovattibr](https://github.com/elovattibr/webmixer) 
It was quite a bit work to get it working without its classes and all the other stuff...

So, if You are interested You may take a look at my repo here [https://github.com/Stephanowicz/moOde-alsaWebEQ](https://github.com/Stephanowicz/moOde-alsaWebEQ)

Maybe this gives You an idea on how to optimize the moOde graphics EQ and handling the switching of the mpc output.

Cheers,

Stephan

[image]

moodeaudio, 2021-07-31:
What version of moOde are you running? The most recent versions resume playback after making any changes to eq or other DSP's and they don't rely on switching MPD outputs.

The Web amixer project is pretty cool. I'll look at it when I get some time.

Stephanowicz, 2021-07-31:
well, maybe not the latest...

nevertheless - what I meant: 
-when enabling/disabling the EQ, mpc STOPS playback and restarts after - but this is not necessary: one only has to PAUSE playback and then may continue after switching

-this also includes all other output switching - but as You can see here, You may switch the output instantly with just a break of ~ 1 second in playback

-changing EQ settings are not instant - one has to save them and then may test...

cheers, Stephan

Stephanowicz, 2021-08-24:
updated:
integration of the presets database from moOde
@moment only the left channel values will be stored - the database only seems to be capable to store one value

[image]

Cheers, Stephan

diist, 2021-09-06:
Playing with the Graphical EQ yesterday I found it clunky, I had to `save` then `test` to try out every change; hitting `test` only would undo the current changes. And if it's possible to pause and re-start the playback instead of stopping it and going back to the beginning of the track that would be great!