moode-player/moode #456: When manually play MPD after BT disconnection, volume is set to zero

roizcorp, 2021-11-03:
Env.

Raspi3
USB DAC

Steps to reproduction:

- play MPD
- Connect BT e.g. a phone (notice MPD stops playing)
- Disconnect BT
- Manually Play again MPD (no Resume setting)
- notice the Volume is set to 0
Expected:
Volume is set as before

moodeaudio, 2021-11-03:
What make and model is the USB audio device?

Reboot and post the Moode startup log
```
moodeutl -l
```

roizcorp, 2021-11-03:
`20211103 063224 worker: -- Start
20211103 063225 worker: Successfully daemonized
20211103 063225 worker: Integrity check (passed)
20211103 063226 worker: File check (OK)
20211103 063227 worker: Session vacuumed
20211103 063228 worker: Session loaded
20211103 063228 worker: Debug logging (OFF)
20211103 063228 worker: -- Audio debug
20211103 063228 worker: ALSA cards: (0:b1 | 1:Headphones | 2:C20 | 3:empty
20211103 063228 worker: MPD config: (2:CA DacMagicXS 2.0 | mixer:(PCM) | card:2)
20211103 063228 worker: ALSA mixer actual (PCM)
20211103 063228 worker: ALSA PCM volume set to (amixer: Unable to find simple control 'PCM',0)
20211103 063228 worker: -- System
20211103 063231 worker: Host     (moode)
20211103 063231 worker: moOde    (7.4.1 2021-10-02)
20211103 063231 worker: RaspiOS  (10.6)
20211103 063231 worker: Kernel   (5.4.77-v7+ #1371)
20211103 063231 worker: Platform (Pi-3B+ 1.3 1GB)
20211103 063231 worker: ARM arch (armv7l, 32-bit kernel)
20211103 063231 worker: MPD ver  (0.22.11_p0x3)
20211103 063231 worker: CPU gov  (ondemand)
20211103 063231 worker: USB boot enabled
20211103 063231 worker: File system not expanded yet
20211103 063231 worker: HDMI port on
20211103 063231 worker: -- Network
20211103 063231 worker: eth0 exists
20211103 063231 worker: eth0 wait 0 for IP address
20211103 063234 worker: eth0 wait 1 for IP address
20211103 063237 worker: eth0 wait 2 for IP address
`

moodeaudio, 2021-11-03:
Log output is unreadable.

Post an issue including the audio device make/model and Moode startup log in the support section at our Forum https://moodeaudio.org/forum

closed 2021-11-03 by moodeaudio

roizcorp, 2021-11-03:
20211103 063224 worker: -- Start
20211103 063225 worker: Successfully daemonized
20211103 063225 worker: Integrity check (passed)
20211103 063226 worker: File check (OK)
20211103 063227 worker: Session vacuumed
20211103 063228 worker: Session loaded
20211103 063228 worker: Debug logging (OFF)
20211103 063228 worker: -- Audio debug
20211103 063228 worker: ALSA cards: (0:b1 | 1:Headphones | 2:C20 | 3:empty
20211103 063228 worker: MPD config: (2:CA DacMagicXS 2.0 | mixer:(PCM) | card:2)
20211103 063228 worker: ALSA mixer actual (PCM)
20211103 063228 worker: ALSA PCM volume set to (amixer: Unable to find simple control 'PCM',0)
20211103 063228 worker: -- System
20211103 063231 worker: Host     (moode)
20211103 063231 worker: moOde    (7.4.1 2021-10-02)
20211103 063231 worker: RaspiOS  (10.6)
20211103 063231 worker: Kernel   (5.4.77-v7+ #1371)
20211103 063231 worker: Platform (Pi-3B+ 1.3 1GB)
20211103 063231 worker: ARM arch (armv7l, 32-bit kernel)
20211103 063231 worker: MPD ver  (0.22.11_p0x3)
20211103 063231 worker: CPU gov  (ondemand)
20211103 063231 worker: USB boot enabled
20211103 063231 worker: File system not expanded yet
20211103 063231 worker: HDMI port on
20211103 063231 worker: -- Network
20211103 063231 worker: eth0 exists
20211103 063231 worker: eth0 wait 0 for IP address
20211103 063234 worker: eth0 wait 1 for IP address
20211103 063237 worker: eth0 wait 2 for IP address