moode-player/moode #464: Moodeaudio forum registration is disabled, how to discuss code contributions?

jeroenwalter, 2021-11-28:
Hi

I'm currently adding some functionality to the moodeaudio backend and frontend, that will allow users to create collections of albums, based on (saved) searches and other criteria. 

I'm doing this, because I have a very large library, about 1TB of music with over 50K songs, that I want to access as separate collections. For instance I don't want to see hundreds of jazz and classical albums in the album list when I'm interested in some pop music. The current search functionality is not sufficient for me.

I want to discuss how I best can integrate this in moodeaudio.
I've pushed my recent additions to https://github.com/jeroenwalter/moode/tree/collections so have a look. 
Mind you, it's still a work in progress.

I also have some opinions about the state and maintainability/readability of the source code I want to discuss, so I can hopefully improve some code.

Maybe you also have a todo list that I can look at, maybe there's something I can help with.

However, I don't have a moodeaudio forum account and forum registration is disabled. 

How do I (and other users) get a forum account? Do I need to send an email to somebody? Because I tried the contact form on the forum some days ago, but haven't heard back yet.

moodeaudio, 2021-11-28:
Email tim at moodeaudio dot org with the username you would like to have and I'll create the account and send you an email with the password. You can change the password anytime afterwords. Self-registrations are currently closed due to ongoing spam activity.

If you have a feature to propose, post a thread  and any info, links, screen shots etc, that are applicable in the "Feature requests" sub-forum. Also mention what is missing from or not adequate in the existing Library search/filter capabilities.

Post ideas on code improvements in a separate issue here in Git repo.

-Tim

closed 2021-11-28 by jeroenwalter

monkeypox, 2021-11-28:
Saved (advanced) searches (which would basically be smart playlists) was something we’d talked about when the filtering search setup was implemented so I think your basic idea has merit. The tricky part is how to present them to the user.

If there were only a few they could just be added to the library menu where it’d be most convenient but that’s too limiting.

So how to accommodate more? The easiest thing would be to just make the user access saved searches like any other playlist but that’s kind of yuck. In ancient moode times we talked about adding a (collapsible?) source pane a la programs like iTunes but never got around to it. This would expose playlists to all views which would be nice. I think it could basically just replace the library menu and have that function added to the top of the source pane.

I agree that moving discussion to the forum where you can bounce ideas off the users and get their feedback seems like a good idea, I’m curious to see what your vision for this is.

jeroenwalter, 2021-11-28:
Before I'm ready to discuss this on the forum, I collected some thoughts on the wiki of my fork.

[https://github.com/jeroenwalter/moode/wiki/Collections](https://github.com/jeroenwalter/moode/wiki/Collections)

I'm currently working on a proof on concept, so I can gather more knowledge about the moode source code and gain more insight for a discussion.

moodeaudio, 2021-12-01:
Collections seem similar to SQL Views. I think most of functionality can be achieved with the existing search/filtering code by allowing the 'OR' operator plus the ability to save searches as named items (Views) i.e., libcache_viewname.json. There would prolly need to be a SQL table that maintained the name of the view and its associated search criteria. This would allow users to select a particular view from the list, delete a view, regenerate a view or all views, etc.

Playlists and Favorites in moOde on the other hand are very basic and there have been many user requests to improve this functionality. Currently the only way to maintain Playlists is via the Queue. There is no "Add to playlist" option in the Library and Playlists can't be edited directly.

Favorites by default are added to a playlist named "Favorites". There can be multiple Favorites playlists each with different names. Names are set via ellipsis menu ... "Set favorites". This is not really that obvious.

-Tim