moode-player/moode #466: add CM4 stanza to config.txt

TheOldPresbyope, 2021-11-30:
The USB ports are disabled by default on the CM4 and must be enabled before moOde can play through a USB DAC. The CM4 documentation would have us add `dtoverlay=dwc2,dr_mode=host` to /boot/config.txt to achieve this.

Not so well documented is the fact that enabling USB OTG by way of the XHCI controller on the CM4 is a better solution. (cf. [USB audio crackling on Compute module 4](https://github.com/raspberrypi/linux/issues/4314)).

Just as Raspberry Pi OS bullseye does, we can accomplish this by adding the following conditional stanza to /boot/config.txt

---snip---
[cm4]
otg_mode=1

[all]
---snip---

This is working for me with moOde 7.6.0(e.g., buster), a CM4 on a Raspberry Pi Compute Module 4 IO Board, and several USB DACs. As expected, it is ignored on other RPi models (maybe dependent on reasonably up-to-date firmware?). Of course, the conditional stanza can be used to load the dwc2 overlay, if desired, instead.

I asked about this otg usage on the RPi Forum. PhilE responded that---off the top of his head---this should work fine for USB host mode; the only reason one might want to revert to the dwc2 overlay is to run the CM4 in USB gadget mode because the XHCI controller doesn't support it. 

Who knows. Maybe someday the docs will be clarified. Or not.

moodeaudio, 2021-11-30:
Nice testing :-)

I'll update config.txt for upcoming r761.

moodeaudio, 2021-12-01:
Commit pushed.

closed 2021-12-01 by moodeaudio