moode-player/moode #470: Stuck on version 7.3.0

obbardc, 2021-12-18:
Hi!
Pressing the `Check for software update` button results in a `Software is up to date` message; but when I login over ssh, the version is `Release 7.3.0 2021-06-25`, which is a few months old.

obbardc, 2021-12-18:
Ah, looking in the logs I got:

```
20210729 140812 updater: Downloading update package r721
20210729 140818 updater: Begin 2021-06-25 update for moOde 7.2.1 2021-05-06
20210729 140818 updater: ** Version check
20210729 140818 updater: ** Step 1-4: Update SQL tables
20210729 140823 updater: ** Step 2-4: Update binaries
20210729 140823 updater: **  Upd 1-4: Install CamillaDSP 0.5.2, GUI 0.7.1, Bckend 0.7.1, Plot 0.5.3
20210729 140941 updater: **  Upd 2-4: Install libcurl3/MPD fix
20210729 140943 updater: **  Upd 3-4: Install librespot 0.2.0
20210729 140943 updater: **  Upd 4-4: Install MPD 0.22.8_p0x3
20210729 140943 updater: ** Step 3-4: Update moOde sources and configs
20210729 140943 updater: ** Step 4-4: Sync changes to disk
20210729 140947 updater: End 2021-06-25 update for moOde 7.2.1 2021-05-06
20210729 140947 updater: Update installed, restart required
```

so I rebooted and checked for updates again and got `Package date: 2021-09-24` which installed fine.

So perhaps the `Software is up to date` message can be misleading?

moodeaudio, 2021-12-18:
In-place updates provide an easy way to upgrade from a given release to the next release. The system must be rebooted after each in-place update is applied.

Depending on project bandwidth for the moOde 8 series effort we may provide a method to upgrade from a given release to the latest release but there is no guarantee this will be included.

closed 2021-12-18 by moodeaudio