moode-player/moode #472: Exception in MDP playing DSD files over PCM-only USB DAC due to an error in mpd.conf

TomsonTung, 2021-12-24:
An configuration error created in MPD result as exception found in mpd log:

[Error]
Dec 24 16:25 : exception: Failed to open "ALSA Default" (alsa); Error opening ALSA device "_audioout"; snd_pcm_hw_params() failed: Invalid argument

[Cause]
The device name is not configured as "_audioout" in mpd.conf.  Due to this error, MPD is not able to query the hw parameters of the device. 

[ Original script in www/inc/playerlib.php] 
	$data .= "audio_output {\n";
	$data .= "type \"alsa\"\n";
	$data .= "name \"" . ALSA_DEFAULT . "\"\n";
	$data .= "device \"_audioout\"\n";
	$data .= "mixer_type \"" . $mixertype . "\"\n";
        ...                                                                                                               

[ Corrected script in www/inc/playerlib.php] 
	$data .= "audio_output {\n";
	$data .= "type \"alsa\"\n";
	$data .= "name \"" . ALSA_DEFAULT . "\"\n";
	$data .= "device \"hw:" . $cardnum . "\"\n";       
	$data .= "mixer_type \"" . $mixertype . "\"\n";
        ...

moodeaudio, 2021-12-24:
Thats not a bug in the code. The ALSA output block should have device set to _audioout which is a virtual ALSA device defined in /etc/alsa/conf.d/ that refers to either hw:x,y or plughw:x,y depending on the Audio Config setting "Output mode" in the ALSA options section.

closed 2021-12-24 by moodeaudio

TomsonTung, 2021-12-24:
Thanks for the quick reply.  But I would like to highlight that the virtual ALSA device setting here does not work with DSD files over PCM-only USB DAC.  MPD was not able to query the virtual device parameter in order to decide how to perform transcoding.  You may see the snd_pcm_hw_params() error in the MPD log in such case. 

I found that if the physical device name is used instead, the DSD->PCM transcoding is performed perfectly.

moodeaudio, 2021-12-24:
Couple things.

1. That MPD error message does not indicate what the cause is and in fact it can result from several different root causes. I'd suggest setting MPD to verbose logging and see if there are any error messages that provide more insight into whats going on.

2. I don't have a PCM-only USB DAC that I could use to try a repro but Its not an issue with PCM-only I2S DAC's. For example below is Audio Info screen from an IQaudIO Pi-DAC+ playing a DSF (DSD128) file.
```
INPUT / OUTPUT
Source			USB/VFAT128/Test_mixed/DSF/2L-45_stereo_01_DSF_5644k_1b.dsf
Encoded at		DSD128, 1 bit, 5.644 MHz Stereo
Decoded to		PCM, 24 bit, 384 kHz, Stereo, 18.432 Mbps
Output rate		24 bit, 384 kHz, Stereo, 18.432 Mbps
Output mode		ALSA Direct (hw)
Output chain		MPD -> hw -> Device

DSP OPERATIONS
Resample rate		Off
Selective rate		Off
Resample quality	Off
CamillaDSP		Off
Chip options		FIR interpolation with de-emphasis, Gain=0dB, Boost=.8dB
Volume type		Hardware (On-chip)

AUDIO DEVICE
Device			IQaudIO Pi-DAC+
Mixer name		Digital
Chip			Burr Brown PCM5122
Interface		I2S
Formats			S16_LE, S24_LE, S32_LE
Platform		Pi-3B+ 1.3 1GB
```

You might want to post your issue in the support section of our Forum to see if anyone can repro it.

Btw what make and model is your USB DAC?

TomsonTung, 2021-12-24:
Thanks again.  I'm using Audioquest Cobalt USB dongle which supports up to PCM 24/96. The following is the audio info when play 16/44 PCM files.

INPUT / OUTPUT
Source 			http://192.168.1.47:9790/minimserver/*/Music/Chlara/2016 In A Different Light/04 True Colours.m4a
Encoded at 		Unknown
Decoded to 		16 bit, 44.1 kHz, Stereo, 754 kbps
Output rate 		24 bit, 44.1 kHz, Stereo, 2.117 Mbps
Output mode 		ALSA Default (plughw)
Output chain 		MPD -> plughw -> Device

DSP OPERATIONS
Resample rate 	Off
Selective rate 	Off
Resample quality 	Off
CamillaDSP 		Off
Volume type 		Hardware (On-chip)

AUDIO DEVICE
Device 			AudioQuest DragonFly Cobalt v1.
Mixer name 		PCM
Interface 		USB
Formats 			S24_3LE
Platform 			Pi-Zero 2 W 1.0 512MB

I attached the MPD log file I attempted to play a DSD file in DSF format
[log.txt](https://github.com/moode-player/moode/files/7775214/log.txt)
.

Before trying GitHub, I try to report this via your forum but I'm not yet a member and the registration function is disabled. 
BTW, Merry Christmas and thanks for making this wonderful software for the community.

Cheers

moodeaudio, 2021-12-25:
Your post shows
```
Output mode ALSA Default (plughw)"
```
Try setting ALSA output mode to Direct (hw)

TomsonTung, 2021-12-25:
Hi Tim, after updated to Direct (hw) per your advice, the DSD->PCM conversion works flawlessly without errors. Once again, thank you for making this software available to the public.  Just submitted my donation today and I hope this project will be successful going forward!  Wish you a Merry Christmas and a wonderful holiday ahead!