moode-player/moode #475: USB Thumb Drive Scanned but not in Library

tokenwizard, 2022-01-03:
I'm running a fresh install of MoOde on a rPi4 via the provided img file. I can successfully play the included SOMA FM Streaming Sources.

I plugged in a USB Thumb Drive with 156GB of music that I was using with a Mopidy setup before. It is formatted EXT4 and after Updating the Library, it looks like it detected the Music and parsed it:

[image]

But when I go back to the main page, even after a Restart of the System, none of the local music files are showing. I'm still only seeing the included Streaming Sources and the one Audio Test file on the SDCard. 

What am I missing. Is there something else that needs to be done to populate USB music files onto the main screen? I also don't see any other way to "Browse" the Music Library. Since I have all my music on this Thumb Drive, I prefer not to use NFS/SMB. I didn't see anything else on the Setup Guide of in the Forums that seems to be relevant to my issues. 

Looking in the Web-based SSH terminal, when I run 'mount' I see the USB Drive did auto-mount with the following parameters:
[image]

And here is a snapshot of the permissions:
[image]

[image]
(Note, after having no success, I tried chown-ing the T and J's Radio folder from the default of shellinabox:audio to pi:audio since it seems the services are running as either the 'pi' user or the 'mpd' user. Also, both 'pi' and 'mpd' users are in the 'audio' group. 

Thanks in advance for any guidance!

________________System Info Pasted Below_____________

		
S Y S T E M   P A R A M E T E R S

moOde release		= 7.6.1 2021-12-17
RaspiOS			= 10.6
Linux kernel		= 5.4.77-v8+ #1371
Platform		= Pi-3B 1.2 1GB
Architecture		= aarch64 (64-bit)
System uptime		= up 8 minutes
Timezone		= America/New_York
Current time		= 2022-01-03 12:49:06

Host name		= TandJ-Radio
Ethernet address	= 192.168.7.157
Ethernet MAC		= b8:27:eb:a1:56:5f
WLAN address		= unassigned
WLAN MAC		= no adapter
WLAN country		= US

SoC identifier		= bcm2837 
Core count		= 4
Kernel timer freq	= 250 Hz
SDCard freq		= 25 MHz
USB boot		= not enabled
Warranty		= OK

Root size		= 15G
Root used		= 20%
Root available		= 12G
Root expand		= expanded
Memory total		= 934 MB
Memory free		= 767 MB
Memory used		= 167 MB
SoC temperature		= 39.7°C
Throttled bitmask	= 0x0
Throttled text		= No throttling has occurred

CPU governor		= ondemand
Onboard WiFi		= Off
Onboard BT		= Off
HDMI output		= Off
LED state		= 1,1
IP addr timeout		= 90 (secs)
Ethernet check		= No
USB auto-mounter	= udisks-glue
SSH term server		= On

[... 268 lines cut ...]
20220103 124052 worker: Stream recorder (n/a)
20220103 124052 worker: -- Music sources
20220103 124052 worker: USB source (T and J's Radio)
20220103 124052 worker: NAS and UPnP sources (none configured)
20220103 124052 worker: -- Other
20220103 124052 worker: USB volume knob (Off)
20220103 124052 worker: Shellinabox SSH started
20220103 124052 worker: USB auto-mounter (udisks-glue)
20220103 124052 worker: LED0 (On)
20220103 124052 worker: LED1 (On)
20220103 124052 worker: Saved MPD vol level (0)
20220103 124052 worker: Preamp volume level (0)
20220103 124052 worker: MPD volume level (0) restored
20220103 124052 worker: ALSA Headphone volume (100%)
20220103 124052 worker: Auto-play (Off)
20220103 124052 worker: Maintenance interval (3 hours)
20220103 124052 worker: Screen saver activation (Never)
20220103 124052 worker: Session permissions (OK)
20220103 124053 worker: Watchdog started
20220103 124053 worker: Ready
20220103 124832 watchdog: Info: Reducing PHP fpm worker pool

	

I

moodeaudio, 2022-01-03:
Your best bet is to post the issue in the Support section of our Forum. Our Git repos are mainly for developer issues.

closed 2022-01-03 by moodeaudio

tokenwizard, 2022-01-03:
I tried to post there, but it says no new registrations are allowed. Since I can't register, I can post.

moodeaudio, 2022-01-03:
Right, I forgot registrations are disabled to curb spam.

Email tim at moodeaudio dot org with your desired userid and I'll create an account for you.

jbrask, 2025-08-13:
I am experiencing a similar issue, also with an ext4 formatted USB. Did you manage to find a solution? If it was posted in the support forum, could you post a link here?

Thanks!