moode-player/moode #477: The moOde forum is closed for new registrations: where can I get help & community support?

ptandler, 2022-01-09:
On the [moOde forum registration page](https://moodeaudio.org/forum/member.php?action=register) there is just a message
`Sorry, but you cannot register at this time because the administrator has disabled new account registrations.`

(When) will it be opened again? (Why is it closed at the moment?)  
Is there any other place to ask for help at the moment?

PS: Thanks for this excellent project! I really love it! I just don't get Bluetooth working on my Pi 4 ...

moodeaudio, 2022-01-09:
Registration is closed due to excessive number of SPAM accounts being registered.

Email tim at moodeaudio dot org with the userid you would like to use on the Forum and I'll setup an account for you with a default password that you can change at your convenience.

-Tim

heman22union, 2022-01-21:
@moodeaudio ,

I will send you an email as well. I saw all the posts on the forms about spam accounts and figured that was the reason. I'm sorry to see the moOde community had to resort to this just to combat the spam. I hope your email isn't spammed now because of it.

Thanks for all the hard work!!
-Heman22union

closed 2022-01-21 by moodeaudio