moode-player/moode #479: Request: Better Playlist Management

tokenwizard, 2022-01-16:
Not sure if this is the best place for this, or if requests are even okay, but since the devs are here, here goes...
 
First, let me say I've been using MoOde regularly for a couple weeks now. SO far I'm loving it. So much faster and more reliable that Volumio or others I've tried, and much easier to configure that a straight MPD or even MoPiDy setup. 

One thing that would make it even better is if there were a better way to manage playlist. While browsing my library, I can click a song and Add/Play/Queue it in the currently tracklist. But there is no option to add to an existing playlist or add/create a new one.

I spent a day putting together a massive playlist of some of my favorite songs in a couple related genres. I did this by just adding all the songs to the tracklist, then saving the tracklist as a new playlist. Then, a few days ago, I wanted to add a handful more songs to the playlist I had created. As best I can tell, there is no easy way to do this. Since there is not "add to playlist" option when you click a song in the library, I decided to clear the tracklist and add all the new songs to the queue, then save that new song tracklist as a playlist with the same name as my existing one. Unfortunately, that just replaced my huge playlist with the new songs. SO I will need to take the time to rebuild that whole playlist now. 

My suggestion would be to add an option when you click a song in the library that is "Add to Playlist" and when you select that, it displays all your existing playlists to choose from as well as a "New Playlist" option. This would bring a pretty typical Music Player feature to MoOde, and, IMHO, make MoOde the de-facto standard for home audio enthusiasts. Also, and perhaps more importantly, there should be a warning before overwriting and existing playlist. 

I am more than willing to help however I can in the endeavor (testing/logs/troubleshooting, etc). 

Thanks for your great work. A donation has been made via the link on your official website. I appreciate all the work devs put into these OSS projects and I hope you are able to make a little money to make the effort worthwhile.

moodeaudio, 2022-01-16:
Right, playlist mgt needs improvement.
Currently you need to first Clear/Add the playlist then manipulate it in the Queue then save it to same name.
One advantage of doing playlist manipulation in the Queue is that it allows moving items around.
A new "Add to playlist" option would always add the item to the end of the playlist which may not be adequate.

tokenwizard, 2022-01-16:
I understand how to work around it for the time being. Also, using a third-part MPD client I can manage playlists within that app (M.A.L.P). Just thought I'd throw in the suggestion/request since I see a lot of users asking for the same thing in the forums. 

Thanks again!

moodeaudio, 2022-01-16:
What features do you want to see in Playlist Management?

tokenwizard, 2022-01-16:
That's pretty much it. That would put MoOde on par with most other media players/MPD clients. 

If there was a way to clear the tracklist without having to click a song and then click Clear/Add, that would be useful, to me at least. 

Personally, I play my playlists on shuffle, so it would not matter if adding a song to an existing playlist adds it at the end. But I can see where others might find that useful. 

And the Add to Playlist options would ideally offer a list of the existing playlists to choose from with a "Create new Playlist" option at the top. Just my $0.02.

monkeypox, 2022-01-16:
> What other features do you want to see in Playlist Management?
> 
> * Add item to Playlist (at end)
> * Add item to playlist (at position)
> * Move item(s) within playlist
> * Save playlist
> * Delete playlist
> * Create playlist from Queue
> * Create new empty playlist

Many moons ago I wrote code to allow interactive editing of a playlist (it went into an “edit mode” on the playback screen where the user selected tracks and then could cut/paste or cancel) but moode didn’t support batched mpd operations at the time and so translating list changes back to mpd would have been ugly and slow. Iirc it wasn’t that much code basically it used the queue list and a class to mark selected items. There was also a way to select to the beginning/end of the list as otherwise it would be much less convenient than the current method. Cut iterated through the list and split the selected items into a temp list that could be pasted later (or not). At the end moode would need to resolve the edited list with mpd, though in retrospect a clear/add operation with the current list would be better than making piecemeal changes. Another thing might be to check for an existing playlist file on save and offering the option to append or overwrite to avoid a situation like the user encountered.

One of the problems with adding “add to playlist” (or anything) there for me was always how large the current menu currently is which I think can make the user go into cognitive overload. So say a “Clear queue” option was added (via the playback & playbar “…” menus I guess?) then moode could maybe get rid of the clear related items on the track/album pop up menus (outside of a configuration option that lets the user choose add or clear behavior universally for Play). And maybe instead of separate “Add” commands moode treated the current queue as one of the options in a new “Add to playlist” dialog. So you’d have:

* Play
* Play next (Add next is basically the same thing, or if constructing a playlist maybe redundant?)
* Add to playlist (brings up dialog)
* Audio info (brings up dialog)

Not having just “Add” might be confusing (too big a change?) or a hassle though idk. And maybe a way to limit the size of the queue so it doesn’t get too ungainly (default to something like 250 tracks with config options for more) might help tame the beast for people who never clear the queue ever.

moodeaudio, 2022-01-16:
I remember that code proposal :-) Maybe we can use it in 8 ?

Right. The context menu is already too long :-0
```
Add
Play
Add next
Play next
Clear/Add
Clear/Play
Audio Info
```

Having a universal Clear queue on the two ... menus might be the way to go. I could see reducing to something like below.  The "Add" options are pretty useful to have right on the menu.
```
Add
Play
Add next
Play next
Add to playlist (dialog)
Audio Info (dialog)
```

closed 2022-07-05 by moodeaudio