moode-player/moode #485: Support for Hifiberry Power Controller

titanium85, 2022-01-26:
Please add support for hifiberry power controller

I'm using Hifiberry AMP2 with power controller. I wasn't able to get that work yet.
Seams that the rotary module on the GUI isn't compatible with the power controller even though the GPIO settings on Moode are changed to GPIO14 and GPIO15.

https://www.hifiberry.com/docs/hardware/power-controller-cabling/
https://github.com/hifiberry/powercontroller

// interrupt pins - uses to signal that a button has been pressed or the controller has been rotated
#define INTERRUPT_PIN1            PIN_PA1   // GPIO4 on Pi
#define INTERRUPT_PIN2            PIN_PF0   // GPIO15 on Pi
#define INTERRUPT_PIN3            PIN_PF1   // GPIO14 on Pi

titanium85, 2022-01-26:
> Peripheral integration is not something we generally do because there are just too many peripherals. Your best bet is to ask the manufacturer for support or post in our Forum and see if another user might be able to help out.

Hi Tim 

Thx for the reply. Unfortunately I wasn’t able to register to Forum, registration is disabled at the moment. 

Kind Regards
Thiago

moodeaudio, 2022-01-26:
We decided to turn off automated registrations due to excessive number of spammer accounts being registered. 

Email tim at moodeaudio dot org with your preferred userid and I'll set up an account for you.