moode-player/moode #489: Feature request: possibility to use buttons with a pull down resistor

ralphcrutzen, 2022-01-31:
I am using a rotary encoder with built-in pushbutton, which is internally connected to 3.3 V instead of GND, so it needs a pull down resistor to work.

To accomplish this, I had to change the `PUD_UP` in `GPIO.setup(sw_1_pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)` to `PUD_DOWN` in /var/www/command/gpio-buttons.py.

I don't now if you are planning for this already, but it would be nice to have this option on the gpio-config.php page.

moodeaudio, 2022-01-31:
I'll add to the TODO list for upcoming moOde 8 series.

What encoder script are you using?

ralphcrutzen, 2022-01-31:
Great, thank you!

By script you mean the parameters on the audio config page? Those are the default values: 100 2 3 23 24.
I'm using this encoder: [https://www.sparkfun.com/products/15141](https://www.sparkfun.com/products/15141)

moodeaudio, 2022-01-31:
Oh ok you are using the built-in rotary encoder driver (python script) thats part of Audio Config to handle encoder rotation and then using the GPIO Button Handler to handle the push button part of the encoder.

ralphcrutzen, 2022-01-31:
Yes, sorry I wasn't clear about the fact that it's only about the pushbutton; not the encoder.

closed 2022-02-17 by moodeaudio