moode-player/moode #490: Bluetooth does not work reliably on RPi4

asomsiko, 2022-02-01:
pi-bluetooth must be updated to at least 0.1.16 to fix inoperable Bluetooth in Moode on RPi4.
https://forums.raspberrypi.com/viewtopic.php?t=304000&sid=3bfd2ecf59c11c9cda406aea7dd1b82e&start=50#p1866271

How to reproduce:
1. Run Mode **7.6.1** on RPi4
2a. Moode fails to connect to BT headphones for audio playback
2b. Pair "Moode Bluetooth" with an Android device. The Bluetooth device paired has no capability listed for Audio playback.

`sudo apt update && sudo apt install pi-bluetooth` fixes the problems

moodeaudio, 2022-02-01:
Here is the version on our upcoming moOde 8 release which is based on Bullseye
```
pi@raspberrypi:~ $ sudo dpkg -l | grep pi-bluetooth
ii  pi-bluetooth                         0.1.18                           all          Raspberry Pi 3 bluetooth

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux
```

moodeaudio, 2022-02-17:
Retest with upcoming moOde 8.

closed 2022-02-17 by moodeaudio