moode-player/moode #493: Moode Player sporadically shuts down.

eldar, 2022-02-15:
Hi! I'm running Moode Audio on Raspberry Pi 4 B. It works really well, however as of late it started sporadically shutting itself down midway through the listening session. Accessing the player through HTTP gives the following error:

```
500 Internal Server Error
nginx/1.14.2
```

I am not able to ssh into the player either. After rebooting it continues working as usual until this happens again at a certain point. I could not register on the forum, so posting it as an issue here. I checked the logs in `/var/log`, but nothing stands out.

Cheers.

moodeaudio, 2022-02-15:
The symptoms suggest a bad SDCard. Write a fresh image of latest release on a known good SDCard. If symptom still occurs then its something external to moOde software.

I normally don't do any support in Git issues, only dev stuff. if you would like a Forum account  email tim at moodeaudio dot org with your preferred userid and I'll set one up for you.

closed 2022-02-15 by moodeaudio

Stooovie, 2022-03-05:
Happens to me with a brand new SD card as well. Card is verified for errors.