moode-player/moode #494: moodeutl and moodeutl help - modest changes

TheOldPresbyope, 2022-02-24:
**1. Clarification issue**

moodeutl says this about its export / import options

```
 -e [filename]	Export settings to moodecfg.ini
 -i		Import moodecfg.ini
```

It isn't clear to first-time users that the exported file will be in the current directory by default and have the name `moodecfg-<time stamp>.ini` by default or `filename` if it is given.

It isn't clear either that the imported file is exactly `/boot/moodecfg.ini` (although they'll discover that quickly enough when they try it).

**2. Good housekeeping issue**

Since the user pi has unrestricted root permissions (via `sudo`) anyway this point may be moot but `moodeutl -e` can write anywhere in the filesystem if executed as root. That's just not good practice. I believe it should strip a given `filename` to its basename and extension, if any, before using it.

moodeaudio, 2022-02-24:
I'll update the help and for (2) I can simply remove the [filename] option from -e. I first have to check if "moodeutl -e [filename]" is used in another part of the code base.

moodeaudio, 2022-02-24:
It's used in backupmanager.py
```
 system('moodeutl -e ' + BackupManager.MOODECFGINI_TMP)
```

It's prolly not very likely that it would be used in such a way as to cause some other file to be accidentally overwritten. More worrisome are things like sudo rm -rf or sudo dd :-0

moodeaudio, 2022-02-28:
The help text was updated in this commit
https://github.com/moode-player/moode/commit/e561e7690618e66d77924713464318cbb2d27f30